nanoframework/Home #622: JsonConvert.DeserializeObject generate error with array short in an object

andceoynnova, 2020-06-03:
### Details about Problem

**nanoFramework area:** (C# code )

**VS version:** ( 2019 )

**VS extension version:** 2019.2.0.1

**Target:** ESP32

**Device capabilities output:**
System Information
HAL build info: nanoFramework running @ ESP32
  Target:   ESP32_WROOM_32
  Platform: ESP32

Firmware build Info:
  Date:     Jun  1 2020
  Type:     MinSizeRel build with IDF v3.3.1
  Version:  1.4.0.207
  Compiler: GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

AppDomains:

Assemblies:
  esp32_nf_Orved, 1.0.0.0
  Windows.Devices.Wifi, 1.2.0.0
  System.Net.Http, 1.2.2.0
  nanoFramework.System.Collections, 1.0.1.0
  nanoFramework.Hardware.Esp32, 1.2.1.0
  Windows.Devices.SerialCommunication, 1.2.1.0
  nanoFramework.Runtime.Events, 1.4.2.0
  nanoFramework.Json, 2.0.0.0
  nanoFramework.System.Text, 1.0.0.0
  mscorlib, 1.7.3.0
  Windows.Storage, 1.2.0.0
  nanoFramework.Runtime.Native, 1.4.0.0
  System.Net, 1.4.1.0
  Windows.Storage.Streams, 1.8.2.0

Native Assemblies:
  mscorlib v100.4.10.0, checksum 0x9D93AC29
  nanoFramework.Runtime.Native v100.0.8.0, checksum 0x2307A8F3
  nanoFramework.Hardware.Esp32 v100.0.7.1, checksum 0x1B75B894
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.7.1, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.4, checksum 0x39DA4F21
  System.Net v100.1.3.1, checksum 0x1118F266
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  Windows.Devices.SerialCommunication v100.1.1.1, checksum 0x82260711
  Windows.Devices.Spi v100.1.4.1, checksum 0x59B5BFC3
  Windows.Devices.Wifi v100.0.6.1, checksum 0xDF2FD922
  Windows.Storage v100.0.2.0, checksum 0x5160A7B6

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe436c  0x0001bc00
  FLASH 0x08000000  0x00800000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x180000     nanoCLR
      1    0x00190000       1      0x1B0000     Deployment
      2    0x00340000       1      0x100000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00340000    0x100000 (1024kB)   Configuration
  0x00010000    0x180000 (1536kB)   nanoCLR
  0x00190000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty

### Description

If a object have an array of ushort or short DeserializeObject launch an exception.

### Other suggested things

### Expected behaviour

Run with all array type.

#### Sample Project
```
public class ChildClass
    {
        public int one { get; set; }
        public int two { get; set; }
        public int three { get; set; }
        public DateTime four { get; set; }

        public override string ToString()
        {
            return $"ChildClass: one={one}, two={two}, three={three}, four={four.ToString()}";
        }
    }
    public class TestClass
    {
        public int i { get; set; }
        public string aString { get; set; }
        public DateTime Timestamp { get; set; }
        public DateTime FixedTimestamp { get; set; }
        public short[] intArray { get; set; }
        public string[] stringArray { get; set; }
        public ChildClass child1;
        public ChildClass Child { get; set; }
        public object nullObject { get; set; }
        //public float nanFloat { get; set; } //<- fails on serialization
        public float f { get; set; }
        public bool b { get; set; }
    }
var test = new TestClass()
            {
                aString = "A string",
                i = 10,
                Timestamp = DateTime.UtcNow,
                FixedTimestamp = new DateTime(2020, 05, 01, 09, 30, 00),
                intArray = new[] { (short)1, (short)3, (short)5, (short)7, (short)9 },
                stringArray = new[] { "two", "four", "six", "eight" },
                child1 = new ChildClass() { one = 1, two = 2, three = 3 },
                Child = new ChildClass() { one = 100, two = 200, three = 300 },
                nullObject = null,
                //nanFloat = float.NaN,
                f = 1.2345f,
                b = true
            };
            var result = JsonConvert.SerializeObject(test);
            Debug.WriteLine($"Serialized Object: {result}");
            var dserResult = (TestClass)JsonConvert.DeserializeObject(result, typeof(TestClass));
            Debug.WriteLine("After Type deserialization:");
            Debug.WriteLine(dserResult.ToString());
```

### Screenshot
[image]

networkfusion, 2020-06-04:
I am not sure how easy this will be to achieve as shorts are pretty much the same as int's when it comes to serialization / deserialization.

Here is a bit more debug output:

```          PopulateObject() -      memberProperty.Value is JObject
                    PopulateObject() -  Start - 
                    PopulateObject() -  rootToken is a JsonObjectAttribute  rootType: ChildClass   rootPath: /Child
                    PopulateObject() -  create rootInstance from rootType: ChildClass using GetConstructor() & Invoke()
                    PopulateObject() -  rootInstance created.  rootInstance.GetType(): ChildClass
                    PopulateObject() -  Entering rootObject.Members loop 
                    PopulateObject() -  Process rootObject.Member
                    PopulateObject() -      memberProperty.Name:  four 
                    PopulateObject() -      memberProperty.Value is JValue
                    PopulateObject() -      attempting to set rootInstance by invoking this member's set method for properties  or  SetValue() for fields
                    PopulateObject() -      successfully initialized member four  to  0 
                    PopulateObject() -  Process rootObject.Member
                    PopulateObject() -      memberProperty.Name:  one 
                    PopulateObject() -      memberProperty.Value is JValue
                    PopulateObject() -      attempting to set rootInstance by invoking this member's set method for properties  or  SetValue() for fields
                    PopulateObject() -      successfully initialized member one  to  100 
                    PopulateObject() -  Process rootObject.Member
                    PopulateObject() -      memberProperty.Name:  three 
                    PopulateObject() -      memberProperty.Value is JValue
                    PopulateObject() -      attempting to set rootInstance by invoking this member's set method for properties  or  SetValue() for fields
                    PopulateObject() -      successfully initialized member three  to  300 
                    PopulateObject() -  Process rootObject.Member
                    PopulateObject() -      memberProperty.Name:  two 
                    PopulateObject() -      memberProperty.Value is JValue
                    PopulateObject() -      attempting to set rootInstance by invoking this member's set method for properties  or  SetValue() for fields
                    PopulateObject() -      successfully initialized member two  to  200 
          PopulateObject() -  Returning rootInstance
          PopulateObject() -      successfully initialized member Child to memberObject
          PopulateObject() -  Process rootObject.Member
          PopulateObject() -      memberProperty.Name:  aBoolean 
          PopulateObject() -      memberProperty.Value is JValue
          PopulateObject() -      attempting to set rootInstance by invoking this member's set method for properties  or  SetValue() for fields
          PopulateObject() -      successfully initialized member aBoolean  to  True 
          PopulateObject() -  Process rootObject.Member
          PopulateObject() -      memberProperty.Name:  aShort 
          PopulateObject() -      memberProperty.Value is JValue
          PopulateObject() -      attempting to set rootInstance by invoking this member's set method for properties  or  SetValue() for fields
    ++++ Exception System.Exception - CLR_E_WRONG_TYPE (1) ++++
    ++++ Message: 
    ++++ nanoFramework.Test.TestClass::set_aShort [IP: 0000] ++++
    ++++ System.Reflection.MethodBase::Invoke [IP: 0000] ++++
    ++++ nanoFramework.Json.JsonConvert::PopulateObject [IP: 03de] ++++```

@skigrinder is this something you would be willing to look at?
i.e. `        public short aShort { get; set; }`
I guess we should also have tests for float arrays also...
i.e. ```        public short[] shortArray { get; set; }
        public float[] floatArray { get; set; }```

skigrinder, 2020-06-06:
Hi Robin,
Sorry, been buried at work.
I can look at adding support for additional types to the Json lib.
Looks like you've made some changes recently and I don't want to collide 
with your efforts.
If you don't have any work going on at the moment, I'll do a fork and 
get started.
Please let me know.
Thanks,
    Jim

networkfusion, 2020-06-06:
Hi Jim, the recent change was only to include the new types to the test project. Please feel free to fork and carry on. Everything has been merged (no PR's open)

josesimoes, 2020-07-02:
Any progress on this?

skigrinder, 2020-07-02:
Hi Jose,
Hoping to have some time this holiday weekend to work on this.
Thanks,
    Jim

networkfusion, 2020-07-09:
fixed in https://github.com/nanoframework/lib-nanoFramework.Json/pull/23

closed 2020-07-09 by nfbot