nanoframework/Home #623: STM32F411EDISCOVERY can't use ST_STM32F411_DISCOVERY's image ?

maikebing, 2020-06-13:
I'm download https://bintray.com/nfbot/nanoframework-images-community-targets/ST_STM32F411_DISCOVERY/_latestVersion  and Program to  STM32F411EDISCOVERY , the VS2019's Device Explorer can't found anything ! 
Did I do anything less?

piwi1263, 2020-06-14:
Hi @maikebing you have to connect a serial -> usb adapter on USART2. 

PA2 for USART2_TX, PA3 for USART2_RX and a GND wire should do.

If not done already, make sure the board is the only connected and perform a target update as:
[image]
 

It shows in VS 16.6.2 after power cycle of the board:
[image]

maikebing, 2020-06-16:
Thank you so much.

closed 2020-06-16 by maikebing