nanoframework/Home #630: ESP32 - Gpio repeated behaviour

valoni, 2020-06-19:
### Details about Problem

**Target:** ESP32

**Firmware image version:** Stable 1.4.2.1 (evend latest develop)

A clear and concise description of what the bug is.

### Detailed repro steps so we can see the same problem

source code : 

```csharp

using System;
using System.Diagnostics;
using System.Threading;
using Windows.Devices.Gpio;
using nanoFramework.Runtime.Native;

namespace BlinkSample
{
    public class Program
    {
       static GpioPin Userled;
        static GpioPin UserBtn;
		
        public static void Main()
        {
            Debug.WriteLine("--------------------------------------------");

            nanoFramework.Runtime.Native.GC.EnableGCMessages(true);

            Debug.WriteLine($"{ SystemInfo.TargetName } running on { SystemInfo.Platform }.");

            nanoFramework.Runtime.Native.GC.Run(false);

            Debug.WriteLine("--------------------------------------------");

            Debug.WriteLine("--------------------------------------------");
            Debug.WriteLine("Welcome ESP32 Board");
            Debug.WriteLine("--------------------------------------------");

            Userled = GpioController.GetDefault().OpenPin(2);
            Userled.SetDriveMode(GpioPinDriveMode.Output);

            UserBtn = GpioController.GetDefault().OpenPin(0);
            UserBtn.SetDriveMode(GpioPinDriveMode.InputPullUp);
            UserBtn.ValueChanged += UserBtn_ValueChanged;

            int cnt = 0;

            while (true)
            {
                Userled.Write(GpioPinValue.High);
                Thread.Sleep(125);

                Userled.Write(GpioPinValue.Low);
                Thread.Sleep(125);

                Debug.WriteLine("Cnt = " + cnt.ToString());
                cnt++;

                Debug.WriteLine(">> " + cnt.ToString() + " free memory: " + nanoFramework.Runtime.Native.GC.Run(false) + " bytes");

            }
        }

        private static void UserBtn_ValueChanged(object sender, GpioPinValueChangedEventArgs e)
        {
            if (e.Edge == GpioPinEdge.FallingEdge)
                Debug.WriteLine("Button Clicked");
            else
                Debug.WriteLine("Button Unclicked");
        }

       
    }
}
```

**behaviour on GpioPinValueChangedEventArgs event trigger unexpected.
and repeat click/unclick multiple time ?? (if button pressed once)** 

>> 214 free memory: 104088 bytes
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Unclicked
Button Clicked
Button Clicked
Button Unclicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Clicked
Button Unclicked
Button Clicked
Button Unclicked
Button Clicked
Button Clicked
Button Unclicked
Button Clicked
Button Unclicked
Button Clicked
Button Unclicked
Button Clicked
Button Unclicked
Button Unclicked
Button Unclicked
Button Clicked
Cnt = 214

### Expected behaviour

Button Clicked (once)
after release button - Button Unclicked (once)

Dweaver309, 2020-06-19:
I can confirm this bug I get the same results using the sample code.

AdrianSoundy, 2020-06-20:
If you are using physical buttons you must add a DebounceTimeout  as the buttons bounce and create multiple events for every click.

```
UserBtn.DebounceTimeout = new TimeSpan(0, 0, 0, 0, 50);
```

The thing that seems wrong is you get  Unclicked followed by another Unclicked  events which is wrong.

valoni, 2020-06-20:
after your suggestion 

i did

```csharp
            UserBtn = GpioController.GetDefault().OpenPin(0);
            UserBtn.SetDriveMode(GpioPinDriveMode.InputPullUp);

            **UserBtn.DebounceTimeout = new TimeSpan(0, 0, 0, 0, 50);**
            UserBtn.ValueChanged += UserBtn_ValueChanged;
```

click work well now , so i belive it just need to be documented this case if used ESP32 ..

closed 2020-06-20 by AdrianSoundy