nanoframework/Home #631: Interfaces aren't inherited

adaslesniak, 2020-06-23:
### Details about Problem

**nanoFramework area:** ( Visual Studio extension | Other tools )

**VS version: 2017 v 15.9.24**

**VS extension version: 2017.2.0.17**

### Description

Information about implemented interfaces isn't kept.

### Detailed repro steps so we can see the same problem

1. Everything is clearly shown in attached example:  https://github.com/adaslesniak/nf_bugs

2. Code doesn't find that object is implementing some interface. The same instance once returns true (that it is IInterface) and other times not, depending to what we cast it before. Couldn't reproduce it if everything was in one project, only happened if Interface and BaseClass were defined in separate project. (all projects are in the same solution linked above)

...

### Expected behaviour
if MyClass is IInterface
then (myClass as object) is IInterface should be true

josesimoes, 2020-07-02:
@adaslesniak can't use the sample project you've added. It's missing the "SourceLibrary" project.

josesimoes, 2020-07-06:
@adaslesniak would like to have this fixed. Care to update the test project as requested?

adaslesniak, 2020-07-07:
Yes, will test today.

adaslesniak, 2020-07-07:
No, it does not work with newest VS 2017 extension. 
I fixed  Test repository so it contains other project and refactored it for clarity:
there are 2 test one: 
1 - does it find underlying type - passes
2 - does it find interfaces of underlying type - does not pass

josesimoes, 2020-07-07:
@adaslesniak I haven't asked for you to test with the new extension. I've request to UPDATE the test solution because it's **MISSING** the referenced library. Please add that.

adaslesniak, 2020-07-07:
That's done.

closed 2020-07-08 by josesimoes