nanoframework/Home #637: Debug.Writeline prevents re-programming

FraserElectronics, 2020-08-11:
### Details about Problem
I have a tight loop where I am calling Debug.Writeline to output debugging information.
I now find that once I have programmed and run the firmware I can no longer re-program my firmware from Visual Studio. Although it still detects the target in Device Explorer it says "No response on COM8".
The only solution is to do a full chip erase and reprogram the bootloader and CLR.

**Target:**
ST Nucleo144 F746ZG

**Firmware image version:**
1.4.590.0

Worked before? If so, with which nanoFramework image version:
Don't know.

**Device capabilities output:**

### Description

Suspect that continuously writing to debug output if preventing other code from running or causing something to timeout ?

### Detailed repro steps so we can see the same problem

Here is the code that causes problems:
```csharp
            long ticksnow;

            while ( true )
            {
                //
                // If we see a gap in the incoming serial data of 1.5 character (byte) times
                // then we will attempt to process what we have received so far.
                //
                ticksnow = DateTime.UtcNow.Ticks;
                Debug.WriteLine( ( ticksnow - _lastIncomingDataTick ).ToString() );
                if ( ( ticksnow - _lastIncomingDataTick ) > ( _halfCharTickTime * 3 ) )
                {
                    lock ( _incomingFrame )
                    {
                        if ( _incomingReceived > 0 )
                        {
                            ProcessModbusFrame();
                            _incomingReceived = 0;
                        }
                    }
                }
            }
```

### Other suggested things

### Expected behaviour

I expect to be able to rebuild and reprogram my code endlessly without having to reflash the whole device.

### Screenshot

### Additional context

### Make an effort to fix the bug

josesimoes, 2020-08-11:
Is the above running on main() or on its own thread?

FraserElectronics, 2020-08-11:
It is running in its own thread.

Andy

steffalk, 2020-08-11:
Hello @FraserElectronics, could it be that you lock(_incomingFrame) nested? Aka something like lock(_incomingFrame) {... lock(_incomingFrame) {...} ...}? Even if both locks happen on separate threads - so you suffer from some deadlock?
Best Regards, Stefan

FraserElectronics, 2020-08-12:
It happens even if I am not sending data so _incomingReceived is always zero. Also, I see the debug text scrolling on screen which means I don’t have a deadlock on my lock.

Andy

steffalk, 2020-08-12:
So should we be able to reproduce the issue just by using while (true) {Debug.Writline("something");}? On what kind of board and firmware did that occur please?

FraserElectronics, 2020-08-12:
It is running on the Nucleo-F746ZG

Andy

josesimoes, 2020-09-07:
Cant' reproduce this... it must be something in your code.

Please attach a project with minimal code that reproduces the behaviour you're describing.

closed 2020-11-14 by stale[bot]