nanoframework/Home #638: Issue with Socket connections on Nucleo-F746ZG

FraserElectronics, 2020-08-23:
### Details about Problem
Unable to connect to a TCP / IP port using Socket class.

**Target:**
Nucleo-F746ZG

**Firmware image version:**
System Information
HAL build info: nanoCLR running @ ST_NUCLEO144_F746ZG
  Target:   ST_NUCLEO144_F746ZG
  Platform: STM32F7

Firmware build Info:
  Date:     Jul 13 2020
  Type:     MinSizeRel build with ChibiOS v19.1.4.1
  Version:  1.4.590.0
  Compiler: GNU ARM GCC v9.3.1

**Device capabilities output:**

### Description

In the example code below, the code never returns from listener.Accept() when using a Telnet client to connect to the board on the DHCP assigned IP address and selected port.

```csharp
IPEndPoint localEndPoint = new IPEndPoint( IPAddress.Any, ListeningOnPort );

Socket listener = new Socket( localEndPoint.AddressFamily, SocketType.Stream, ProtocolType.Tcp );

listener.Bind( localEndPoint );

listener.Listen( 100 );

Debug.WriteLine( $"Waiting for an incoming connection on port {ListeningOnPort}.." );

_clientSocket = listener.Accept();

Debug.WriteLine( "Connected to a client" );
```

### Detailed repro steps so we can see the same problem

An example solution can be found in GitHub here: [NFTCPServer](https://github.com/FraserElectronics/NFTCPServer.git)

### Other suggested things

### Expected behaviour

Would accept Socket.Accept() to return on a client connection.

### Screenshot

### Additional context

### Make an effort to fix the bug

Attempt to submit a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) that fixes the bug. Include in this PR a test that verifies the fix. If you were not able to fix the bug, a PR that illustrates your partial progress will suffice.

networkfusion, 2020-11-30:
@FraserElectronics have you had any more sucess with this on later versions? there was quite a big overhaul of the network stack when we changed to chibios 20.3.x.

FraserElectronics, 2020-11-30:
No, I have not had a chance - I will try and dig out my Nucleo-F746ZG and give the latest version a try.

Andy

josesimoes, 2021-04-13:
Closing by lack of feedback. Reopen if needed.

closed 2021-04-13 by josesimoes