nanoframework/Home #645: Update nuspec files for all class libraries

nfbot, 2020-09-25:
Need to add project logo as package icon and add a new variable in the description.

- [ ] add project logo
- [ ] update license to MIT
- [ ] update owners and authors
- [ ] add native checksum variable

Basically follow this PR: nanoframework/lib-nanoFramework.Networking.Sntp#82

networkfusion, 2020-10-30:
Since `nanoframework.Json` only required a change of license, this was committed directly.

josesimoes, 2020-12-31:
In progress!

networkfusion, 2021-03-13:
In progress!

josesimoes, 2021-06-09:
In progress

josesimoes, 2021-08-18:
almost done!

networkfusion, 2021-09-10:
@josesimoes are you aware of any libs this still affects?

josesimoes, 2021-09-11:
> @josesimoes are you aware of any libs this still affects?

I've been clearing them as I have to touch the projects for other reasons. Don't think they are many left....

josesimoes, 2021-11-25:
DOOOOOOOONNNEEEE!!! 😅

closed 2021-11-25 by josesimoes