nanoframework/Home #649: Provide HTTP web server as NuGet package

nfbot, 2020-09-25:
Ideas:
- Dynamic content (kind of aspx)
- Strong inclination to provide REST service (IoT)

Some pointers:
https://github.com/michaelschwarz/NETMF-Toolkit
https://github.com/microcompiler/microserver
https://github.com/Rinsen/WebServer

Related issue: nanoframework/Home#474

danielmeza, 2020-10-09:
I believe that static content will be a nice first versión, this way we can put any client side application on it. (using an external storage), the second step will be to create an web api framework to handle the http request.

Ellerbach, 2020-10-09:
Very very very old and I built this for .NET MF :-) 
https://archive.codeplex.com/?p=netmfwebserver
In short a simple web server which can serve static files and some dynamic way of working with some controllers (functions) build for .NET Microframework. Worked perfectly for years on a Netduino serving my sprinkling system.

danielmeza, 2020-10-10:
@Ellerbach You are a issue killer hahah thanks!

Ellerbach, 2020-10-13:
> @Ellerbach You are a issue killer hahah thanks!

@danielmeza and @josesimoes, this is quite old code, really, from very early .NET Microframework time. So it is for sure to be cleaned. I used the same kind of approach at the early UWP time where there was no web server available at all. It's quite basic, quite rustic, efficient (never had issue with it) if you handle a bit properly time outs and things like that. It scales as good as the main underlaying platform scale (so not much for .NET Microframework :-)) but at the end of day, it does the job for few simultaneous connections.
I really really really don't have the time to clean it those days. If someone can that's great. If not, then this will wait for more clear sky in terms of time on my side :-)

Ellerbach, 2020-10-13:
@danielmeza, I had a look anyway :-) What needs to be adjusted is little:
- Debug.Print to Debug.WriteLine
- Some of the  NetworkInterface property in the Debug line IPAddress to IPv4Address
- FileStream is the problematic part, the Windows.Storage component should be used instead. There is the sample which explain how to access file:
  - first make sure the SD Card is mounted: https://github.com/nanoframework/Samples/blob/master/samples/Storage/MountDevices/Program.cs
  - Then you open the file and get the bytes: IBuffer readBuffer = FileIO.ReadBuffer(myFile);
https://github.com/nanoframework/Samples/blob/8b8479cde50001886f3c77492d3e6149ab6d7c48/samples/Storage/FileAccess/Scenario4_WriteAndReadBytesInAFile.cs#L32
And that's about it!

Something that may be missing is a property for the root folder where the files will be stored. You can find that dynamically as well by getting the list of files. 
Again, the example is super clear and you won't get lost :-)
Happy to help!

Ellerbach, 2020-10-14:
@danielmeza and @josesimoes, I spent few hours at the end of my work day to port the WebServer on nanoFramework.
Result here: https://github.com/Ellerbach/nanoFramework.WebServer
Including a full example with handling static files on a mounted external storage and also basic example of html code generation, and on internal storage as well. 
Quite simple, straight forward and ready to be extended!

Ellerbach, 2020-10-19:
@danielmeza I ported the code to nanoFramework, added simple REST API support, check it there: https://github.com/Ellerbach/nanoFramework.WebServer
Feedback welcome!

Ellerbach, 2020-11-05:
And if anyone is interested and following this thread, the WebServer is now in an advance version supporting all what was asked and I've migrated to https://github.com/nanoFramework/nanoFramework.WebServer. 
Soon a nuget to play with :-)

josesimoes, 2021-01-06:
Library is now available.
Sample project on [Hackster.io](https://www.hackster.io/ellerbach/net-nanoframework-rest-api-and-web-server-3c2e9e).

closed 2021-01-06 by josesimoes