nanoframework/Home #651: Migrate IDF to v4 on ESP32

danielmeza, 2020-10-08:
# **IDF 4 Migration**

Hi every one, this issue is to track and place all context for the progress on the EDF v4 migration for ESP32 target platform.

**nanoFramework area:** (Hardware/ESP32_WROOM_32)

### Is your feature request related to a problem? Please describe.
The migration of IDF 3 to 4 is a big task to accomplish in order to get access to the new api set and implement missing key parts or ESP32 platform (BLE) 

### Describe the solution you'd like
Migrating the IDF involves several steps. Here's a general overview of the process.

- [ ] Recompiling the libraries.
- [ ] Updating the various references to the new IDF version in the CMake files and PS scripts
- [ ] Adjust calls to the IDF functions/API, if needed.

### Describe alternatives you've considered
Use the current API for BLE on v3, but due to the size and complexity it is not a viable solution see [this ](https://github.com/nanoframework/Home/issues/476#issuecomment-668067882)

### Additional context
As background discussion you can se [this issue](https://github.com/nanoframework/Home/issues/476#issuecomment-668067882).

### Submit a PR with the implementation
PR In progress. for

AdrianSoundy, 2020-10-08:
I have already looked at doing this and there are various issues to overcome.
The first issue is they have changed the way the TCP stack works so needs some extra work/thought.

Also it would be better to start again with the current cmake project to use the facilities the ESP32 IDF now provides for including the IDF into an existing project instead of using pre-compiled binaries.  But this could be done in separate issue.

danielmeza, 2020-10-08:
Fot the TCP I guess your are talking about [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/tcpip_adapter_migration.html?highlight=tcp).

We can start to migrate the CMake to merge it with the IDF without migrate the TCP stack?

dagid4, 2021-02-08:
Any thoughts about when this could be done?

I'm deciding whether to use nanoframework for my app, but without bluetooth I cannot do it. Thanks for any response.

lancefi, 2021-04-24:
Any update on this and BLE support?

josesimoes, 2021-04-24:
Interestingly there were rumors about this matter during the past week. It's still on slow motion.

AdrianSoundy, 2021-04-26:
Because the IDF to 4.x is going to take some time to complete as i have decided to first do a migration to v3.3.4 first using existing nuild system so that nimble BLE can be enabled.

AdrianSoundy, 2021-04-26:
Changes:-

Update Install scripts
-

AdrianSoundy, 2021-04-26:
Build Libs
- Enabler Nimble BLE

AdrianSoundy, 2021-04-26:
Check Patched versions of Lwip TCP stack changes for Socket API

AdrianSoundy, 2021-04-26:
Update 
https://docs.nanoframework.net/content/esp32/build_ESP32_libraries.html

AdrianSoundy, 2021-04-27:
Build issue :- TODO
**Adc driver**
The adc_power_on() has been deprecated and replaced by adc_power_acquire() but this should have a matching adc_power_release()

AdrianSoundy, 2021-04-29:
Needed to update linker files otherwise the elf -> bin util would fail with too many(42) segments, max 16 
Fixed issues with missing tcp_isn.c file and updates lwipopts.h

Now image builds and flashes ok
Seen in Visual studio
Tested
- GPIO
- SPI display SD1331
-

AdrianSoundy, 2021-04-29:
TODO:-
Need to do a test that Nimble bluetooth can be compiled in.
Add dummy assembly with init code.
Add Parameter to enable it

AdrianSoundy, 2021-05-06:
Seems when you include the Bluetooth the iRam0 segment overflows.
This area is 128K and is used for running code in ram for faster execution and stop swapping of flash to cache in interrupts etc.

With bluetooth this area goes over the limit by 10K.  I can't really see anything we can remove as nanoClr only uses a few K.  One of the big users of Iram is when psram is enabled.

We may need to have a version with Bluetooth and no psram
and a separate version with psram and no bluetooth. The latest ESP32 V3 fixes the spiram issue and would allow a full version.

AdrianSoundy, 2021-07-21:
A migration to v3.3.5 has been done first
V4.3 will be done next

closed 2021-10-13 by josesimoes