nanoframework/Home #656: Timeouts do not work in network sockets

MateuszKlatecki, 2020-10-29:
### Details about Problem

**Target:** Any

**Firmware image version:** Latest

### Description

Timeouts in network sockets do not work

### Detailed repro steps so we can see the same problem

1. Deploy program below
```cs
using System;
using System.Diagnostics;
using System.Net;
using System.Net.Sockets;
using System.Threading;

namespace SocketTest
{
    public class Program
    {
        public static void Main()
        {

            var socket = new Socket(AddressFamily.InterNetwork, SocketType.Stream, ProtocolType.Tcp)
            {
                SendTimeout = 5000,
                ReceiveTimeout = 5000
            };

            IPEndPoint ep = new IPEndPoint(IPAddress.Parse("192.168.3.4"), 502);
            socket.Connect(ep);

            var receiveBuffer = new byte[10];
            try
            {
                socket.Receive(receiveBuffer);
            }
            catch (Exception e)
            {
                Debug.WriteLine(e.ToString());
            }

            Thread.Sleep(Timeout.Infinite);
        }
    }
}

```

2. Assuming the server doesn't send anything, 5 seconds after the Receive method is called, a SocketException should be thrown, but it is not.

### Expected behavior

An exception should be properly thrown on a timeout.

josesimoes, 2020-10-29:
Do you know what's causing it?

MateuszKlatecki, 2020-10-29:
Timeout is always set to infinite:
https://github.com/nanoframework/nf-interpreter/blob/859be9a725cbff45fa04d1b45fbb219aec96646b/src/DeviceInterfaces/System.Net/sys_net_native_System_Net_Sockets_NativeSocket.cpp#L688
the timeout parameter (Arg5) is not used

josesimoes, 2020-10-29:
That's the intended behaviour to allow thread execution. Any timeout/error/etc is handled at lwIP layer.
Otherwise all calls would be blocking.

MateuszKlatecki, 2020-10-29:
I know this mechanism - it used in many places. But I think timeout should be set to proper value.  I think timeout is not passed in the method parameter for no reason. See PR: https://github.com/nanoframework/nf-interpreter/pull/1753

closed 2020-10-30 by josesimoes