nanoframework/Home #659: System.Device.I2c hangs at device.write(buffer)

josesimoes, 2020-10-30:
### Details about Problem

**nanoFramework area:** (C# code)

**VS version:** 

**VS extension version:**

**Target:** ESP32 Devkit V4

**Device capabilities output:**

### Description

System.Decice.I2c hangs writing to oled display on esp32.

### Expected behaviour

Application should display text on oled device.

#### Sample Project
[nf_ssd1306.zip](https://github.com/nanoframework/Home/files/5461726/nf_ssd1306.zip)

### Screenshot

cc @palmtreefrb

AdrianSoundy, 2020-10-31:
This is not working in sys_dev_i2c_native_System_Device_I2c_I2cDevice.cpp . 
Although readSpanByte is 0 the if statement is not working so it gets an exception and restarts !
```
        readSpanByte = stack.Arg2().Dereference();
        if (readSpanByte != NULL)
        {
            readBuffer = readSpanByte[SpanByte::FIELD___array].DereferenceArray();
  ```

AdrianSoundy, 2020-10-31:
The call to NativeTransmit with the read parameter as NULL ends up with a non null reference so it tries to pick up read parameter details.
```
        readSpanByte = stack.Arg2().Dereference();
        if (readSpanByte != 0)
        {
            readBuffer = readSpanByte[SpanByte::FIELD___array].DereferenceArray();

  ```

AdrianSoundy, 2020-11-01:
Changes

- Added check for SpanByte Array field == NULL for each platform system.device.I2c ( Read & Write)
- Added TypeSystem entry for m_I2cTransferResult and renamed Windows.Devices.I2c to m_I2cTransferResult_old
- Updated all Windows.Devices.I2C to refer to m_I2cTransferResult_old ( These will be removed later )
- Remove unnessary memcpy at end of ESP32 System.Device.I2c

closed 2020-11-03 by josesimoes

palmtreefrb, 2020-11-03:
Just tested with the latest and it is working as expected. Thanks...