nanoframework/Home #663: CSC compiler error CS7038 "Failed to emit module" when using foreach() on a String

flyingoverclouds, 2020-11-15:
### Details about Problem
**nanoFramework area:** BCL (String class) : C# code failed to compile

**VS** : 2019 16.8.1 (latest update installed)

**VS extension version** : 2019.2.3.5 (lates available version)

**Target** : build/code compilation

**Device capabilities output** : device independant error (compilation step)

**CoreLibrary** : nuget nanoFramework.CoreLibrary v1.9.1-preview.6

### Description
The following code failed to compile 

```csharp
   string s = "HELLO";
   foreach (var c in s)
       Debug.Write(c.ToString());
```
The C# compiler generate a non specic error : 
**CSC : error CS7038: Failed to emit module '_TestForeachString_': Unable to determine specific cause of the failure.**

It appears that using the foreach on a string trigger this behaviour. 

### Detailed repro steps so we can see the same problem

1. Create a new nanoframework app.

2. add the following code in the main() (or anywhere in the code)
```csharp
    string s = "HELLO";
    foreach (var c in s)
        Debug.Write(c.ToString());
```
3. launch the build/Compilation. it will fail.

### Other suggested things
I change the failing code by using .ToCharArray() and it works fine. 

```csharp
    var ls = "HELLO".ToCharArray();
    foreach (var c in ls)
         Debug.WriteLine(c.ToString());
```

### Expected behaviour

The code should compile and should works fine ( displaying the string content char by char)

#### Sample Project

```csharp
using System;
using System.Diagnostics;
using System.Threading;
using System.Collections;
namespace TestForeachString
{
    public class Program
    {
        public static void Main()
        {
            //var ls = "HELLO".ToCharArray(); 
            //foreach (var c in ls)
            //    Debug.WriteLine(c.ToString());

            string s = "HELLO";
            foreach (var c in s)
                Debug.Write(c.ToString());

            Thread.Sleep(Timeout.Infinite);
        }
    }
}

```

### Screenshot
In output windows : 
_1>------ Build started: Project: TestForeachString, Configuration: Debug Any CPU ------
1>CSC : error CS7038: Failed to emit module 'TestForeachString': Unable to determine specific cause of the failure.
1>Done building project "TestForeachString.nfproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========_

josesimoes, 2020-11-15:
Duplicated of #275.

flyingoverclouds, 2020-11-16:
I can accept the reason : avoiding increasing the assembly size. 
But as IEnumerable is defined in mscorlib, why not just returning a NotSupported or PlatformNotSupported exception when the GetEnumerator() is called (with a explicit message). 
Even it is not the best solution, it will be cleaner  that maintaining the compiler crash, and will be compliant with the [NotImplemented exception doc](https://docs.microsoft.com/en-us/dotnet/api/system.notimplementedexception) : 
> _"The .NET Framework also includes two other exception types, **NotSupportedException** and **PlatformNotSupportedException**, that indicate that no implementation exists for a particular member of a type. You should throw one of these instead of a NotImplementedException exception under the following conditions."_

The **NotSupportedException** is already defined in nanoframework mscorlib 1.9.1

josesimoes, 2020-11-16:
Fair enough and makes perfect sense! 

Could you please take care of that and submit a PR with those changes?

closed 2020-11-16 by josesimoes