nanoframework/Home #668: Need to tweak FATFS configuration in ESP32

josesimoes, 2020-12-11:
**nanoFramework area:** ESP32 build

### Is your feature request related to a problem? Please describe.
The System.Io.File API requires that the following configurations are available for FATFS:
FF_USE_STRFUNC  1
FF_USE_FIND 1
FF_USE_MKFS 0
FF_FS_RPATH 2

### Describe the solution you'd like
Considering that the IDF is pre-compiled the above configurations need to be included in the next build so the API can be available to ESP32 targets.

cc @AdrianSoundy

josesimoes, 2020-12-11:
Good candidate to be included in #650.

AdrianSoundy, 2021-04-27:
Doing as part of the IDF V4 (3.3.5) #651

Ellerbach, 2021-06-26:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

Dear bot, this is still valid!

closed 2021-07-12 by josesimoes