nanoframework/Home #669: Multicast Delegate Invocation causes exception under certain circumstance.

andybeetz, 2020-12-16:
### Details about Problem
Calling an event with multiple handlers throws CLR_E_WRONG_TYPE when one of the handlers is a static method on the Program class.

**Target:**
ESP32_WROOM_32

**Firmware image version:**
1.6.1-preview.3

Worked before?:
untested

**Device capabilities output:**
System Information
HAL build info: nanoFramework running @ ESP32
  Target:   ESP32_WROOM_32
  Platform: ESP32

Firmware build Info:
  Date:        Dec 11 2020
  Type:        MinSizeRel build with IDF v3.3.1
  CLR Version: 1.6.1.3
  Compiler:    GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  ESPMulticastDelegateTest, 1.0.0.0
  mscorlib, 1.10.0.0

Native Assemblies:
  mscorlib v100.5.0.3, checksum 0x2F473B03
  nanoFramework.Runtime.Native v100.0.8.0, checksum 0x2307A8F3
  nanoFramework.Hardware.Esp32 v100.0.7.1, checksum 0x1B75B894
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.4, checksum 0x39DA4F21
  System.Net v100.1.3.1, checksum 0x1118F266
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  Windows.Devices.SerialCommunication v100.1.1.1, checksum 0x82260711
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  Windows.Devices.Wifi v100.0.6.1, checksum 0xDF2FD922
  Windows.Storage v100.0.2.0, checksum 0x5160A7B6

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe436c  0x0001bc00
  FLASH 0x08000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x180000     nanoCLR
      1    0x00190000       1      0x1B0000     Deployment
      2    0x00340000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00340000    0x040000 (256kB)    Configuration
  0x00010000    0x180000 (1536kB)   nanoCLR
  0x00190000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty

### Description

Calling an event with multiple handlers throws CLR_E_WRONG_TYPE when one of the handlers is a static method on the Program class.

### Detailed repro steps so we can see the same problem

1. Run attached project on ESP32
[MulticastDelegateTest.zip](https://github.com/nanoframework/Home/files/5701768/MulticastDelegateTest.zip)

### Expected behaviour

Handlers are called without inherent thrown exceptions.

### Screenshot
[image]

### Additional context

When a handler of the Program class is the only registered handler, no exception is thrown.

josesimoes, 2020-12-16:
What is the nF VS extension version that you have installed?

andybeetz, 2020-12-16:
I have version 2019.4.0.31 installed.

AdrianSoundy, 2021-01-08:
Invocation List looks OK, adding and removing works.
Calling each entry in Invocation List also works OK

```
            for (int ctr = 0; ctr < Tick.GetInvocationList().Length; ctr++)
            {
                TickEventHandler outputMsg = (TickEventHandler)Tick.GetInvocationList()[ctr];

                outputMsg(DateTime.UtcNow);

            }
```

Its just the Invoke that doesn't work when multiple entries in Invocation List

AdrianSoundy, 2021-01-08:
The problem happens in CLR_RT_HeapBlock::PerformUnboxing when it tries to unbox object. It checks reference is VALUETYPE and IsBoxed flag set. This doesn't seem to be set for DateTime.

When do same test with a long as argument for event handler it all works.

closed 2021-01-12 by AdrianSoundy