nanoframework/Home #672: Firmware 1.6.X.X shows error System.Net.Security.SslNative::SecureClientInit when having Web server + https request + SD card on ESP32

jefmad, 2020-12-16:
### Details about Problem

**Target:**
ESP32_WROOM_32

**Firmware image version:**
Error is with FW greater than 1.4.0-preview.107

Worked before? If so, with which nanoFramework image version:
1.4.0-preview.107

**Device capabilities output:**
Not relevant, see samples below.

### Description

After firmware version 1.4.0-preview.107, there is a bug that is still present that prevents System.Net and System.Net.Htpp to work correctly when Windows.Storage is also present. The most visible symptoms are 
-	Web server stops to answer request and crashes
-	SSL requests end up with error System.Net.Security.SslNative::SecureClientInit
Conclusion :
FW 1.4.0-preview.107
-	Web server OK when reading Internal Storage
-	Web server OK when NOT reading Internal Storage
-	SSL request OK when reading External Storage
FW 1.6.0-preview.19
-	Web server OK when reading Internal Storage
-	Web server KO when NOT reading Internal Storage
-	SSL request KO when reading External Storage

I created 2 samples so that you can reproduce the issues and dig deeper into the root cause. So far to me, the only “stable version” is 1.4.0-preview.107
Please let me know if you can reproduce the issue I’m dealing about.
So maybe the issue about SSL appeared because of this modification:
Improvements in SSL accept handler [\#62](https://github.com/nanoframework/lib-nanoFramework.System.Net.Http/pull/62)

### Detailed repro steps so we can see the same problem

!! Note that if during tests your device gets the error : System.Net.Security.SslNative::SecureClientInit, the only way to get rid of it is to flash the device. Once this error is obtained, the following test are not accurate anymore, flashing is needed to go on.

Step to reproduce the issues (The two samples are identical, the difference is version of assemblies, I let all the extra assemblies in the project because they were part of my initial project, some are useless in this sample) :

[1.4.107.zip](https://github.com/nanoframework/Home/files/5704859/1.4.107.zip)
[1.6.19.zip](https://github.com/nanoframework/Home/files/5704860/1.6.19.zip)
1. Error with version 1.6.19 (web server and SSL request)
•	Open the solution KO1619.sln
Tried with VSIX 2019.2.1.1, and 2019.2.3.5
•	Flash device (modify your COM port): nanoff --serialport COM5 --fwversion 1.6.0-preview.19 --target ESP32_WROOM_32 –update
•	Modify your wifi info in the code : 
private const string c_SSID = "xxxxx";
private const string c_AP_PASSWORD = "xxxxx";
•	Run the program once in debugging mode, it will stop because not acquiring IP address. Stop debugging
•	Run the program a second time, output should display something like this :
con_C0
con_C1
con_C2
con_A0
con_W0
IP.txt
Already have an IP Address: x.x.x.x
11/03/2020 20:15:10 : ESP32 time is11/03/2020 18:15:10
11/03/2020 20:15:10 : SNTP True
11/03/2020 20:15:14 : Entering Track_IP_Change
11/03/2020 20:15:14 : Start of New loop of Track_IP_Change
11/03/2020 20:15:14 : Entering MakeWebRequest http://icanhazip.com
The thread '<No Name>' (0x1) has exited with code 0 (0x0).
Http response follows
>>>>>>>>>>>>>
11/03/2020 20:15:16 : this is what we got from http://icanhazip.com: xxxxx
Trying to close httpResponse
11/03/2020 20:15:16 : Exiting MakeWebRequest
11/03/2020 20:15:18 : New adress 
11/03/2020 20:15:18 : End of New loop of Track_IP_Change, now sleep for 30 min
11/03/2020 20:15:18 : Update and Stopped sntp

•	Now open a browser and enter the local IP address stated above in the section "Already have an IP Address"
•	See the content displayed in the browser and the output in visual studio
•	Refresh slowly 10 times the page: The program should not crash
•	Check the SSL mark and follow the output in VS : Refresh the web page, and see the beginning of the SSL response
•	Check, uncheck several time. The program does not crash. SSL requests are correctly issued
•	Now, comment the line Actual_IP = FileIO.ReadText(StorageFile.GetFileFromPath(SD.Path + "IP.txt")); 
•	Refresh the web page, it will crash in less than 10 refreshes (the webpage will stop to display, and the output in VS will show a stop / reboot)
•	Now uncomment the line Actual_IP = FileIO.ReadText(StorageFile.GetFileFromPath(SD.Path + "IP.txt"));
•	Comment the line  SD = Internal_Storage;  //*
•	Uncomment the line //            mount_SD();
•	Be sure to have a SD card plugged MOSI :23,MISO 19,CLK:18 and CS on Pin 5 or adapt the code to your SD card configuration
•	If you refresh the web page, it will crash after less than 10 refreshes.
•	Try to check the SSL mark after a single refresh of webpage and program will end up with (if the device crashes before you can click the checkmark, disconnect the device, flash it, disconnect it again) :
Performing Http request to: https://www.google.com/
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (4) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureClientInit [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 003d] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsClient [IP: 000a] ++++
    ++++ System.Net.HttpWebRequest::EstablishConnection [IP: 0234] ++++
    ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 0019] ++++
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++
    ++++ NFApp1.Program::SSL [IP: 0043] ++++

2.Using the version 1.4.107 : All the features work
•	Flash the device with nanoff --serialport COM5 --fwversion 1.4.0-preview.107 --target ESP32_WROOM_32 --update
•	Install the VSIX extension 1.2.0.3
•	Open the solution OK14107.sln
•	Modify your wifi info in the code : 
private const string c_SSID = "xxxxx";
private const string c_AP_PASSWORD = "xxxxx";
•	Run the program once in debugging mode, it will stop because not acquiring IP address. Stop debugging
•	Run the program a second time, output should display something like this :
con_C0
con_C1
con_C2
con_A0
con_W0
IP.txt
Already have an IP Address: x.x.x.x
11/03/2020 20:15:10 : ESP32 time is11/03/2020 18:15:10
11/03/2020 20:15:10 : SNTP True
11/03/2020 20:15:14 : Entering Track_IP_Change
11/03/2020 20:15:14 : Start of New loop of Track_IP_Change
11/03/2020 20:15:14 : Entering MakeWebRequest http://icanhazip.com
The thread '<No Name>' (0x1) has exited with code 0 (0x0).
Http response follows
>>>>>>>>>>>>>
11/03/2020 20:15:16 : this is what we got from http://icanhazip.com: xxxxx
Trying to close httpResponse
11/03/2020 20:15:16 : Exiting MakeWebRequest
11/03/2020 20:15:18 : New adress 
11/03/2020 20:15:18 : End of New loop of Track_IP_Change, now sleep for 30 min
11/03/2020 20:15:18 : Update and Stopped sntp

•	Now open a browser and enter the local IP address stated above in the section "Already have an IP Address"
•	See the content displayed in the browser and the output in visual studio
•	Refresh slowly 10 times the page: The program should not crash
•	Check the SSL mark and follow the output in VS : Refresh the web page, and see the beginning of the SSL response
•	Check, uncheck several time. The program does not crash. SSL request are correctly issued
•	Now, comment the line Actual_IP = FileIO.ReadText(StorageFile.GetFileFromPath(SD.Path + "IP.txt")); 
•	Refresh the web page. The program does not crash compared to version 1.6.19
•	Now uncomment the line : Actual_IP = FileIO.ReadText(StorageFile.GetFileFromPath(SD.Path + "IP.txt"));
•	Comment the line : SD = Internal_Storage;  //*
•	Uncomment the line //            mount_SD();
•	Be sure to have a SD card plugged MOSI :23,MISO 19,CLK:18 and CS on Pin 5
•	Refresh the web page, check the SSL mark, All the features will work compared to version 1.6.19

### Expected behaviour

What is expected is that sample with FW 1.6.0.19 runs without error when performing ssl request as the sample using FW 1.4.0.107

### Screenshot
See samples

jefmad, 2021-02-15:
In order not to be closed after having been marked as stale... Need to test persistent connection and System.Io.FileSystem

josesimoes, 2021-03-19:
Can't reproduce.

Everything seems to be working as expected with fw and libraries released in this date.

Because I have a WROVER KIT, I had to comment out the SPI config lines and replace the SD card mount with this one:
`SDCard.MountMMC(false);`

I even replaced the HTTP files that you have in the project with the HTTP.Client library.

Made  calls in a row to `Perform_SSL_HTTPRequest()`. All good except a parse error from the content.

jefmad, 2021-03-20:
Ok, thanks for the feedback and your tests !  The issue is obtained on ESP32 WROOM-32, not on ESP-WROVER.

After your comment, I thought that SPI config could be the problem, and I tested with another SPI config lines  : exactly the SPI1 config defined here : https://docs.nanoframework.net/content/esp32/esp32_pin_out.html (// Default pins for "SPI1" : Mosi=23  Miso=25  Clk=19) and still the same result : if not mounting SD card via the command SDCard.MountSpi("SPI1", XX); the SSL requests are performed without any problem, as soon as the SD card is mounted, SSL requests fail with the error SecureClientInit.

The parse error are linked to the content and I got them sometimes, it's ok with this. And the http file in the project were present to debug more deeply :-)

**It would be nice if another person with a ESP32 WROOM-32 and an external SD card could test**. I tested on two ESP32 WROOM-32 and it was working before FW 1.4.0-preview.107 and not working anymore... Thanks

josesimoes, 2021-04-13:
Is this still an issue?

jefmad, 2021-04-14:
I have not tested with latest FW, but to me my last comment is still valid : **It would be nice if another person with a ESP32 WROOM-32 and an external SD card could test.** SPI config could be the problem
Is there any improvement in the availability of System.IO for ESP32 ? This could be an option that allow to close this issue

Ellerbach, 2021-04-23:
So I finally manage to reproduce the issue. It's **only** happening with ESP32_WROOM, seems only in wifi.
* issue is yes, related to SDCard in this configuration.
* When mounting the SD Card, a lot of memory is probably allocated and once you want to use SSL Stream, quite some allocated as well
* All up, that *most likely* creates a memory issue somewhere
* The fact that the problem stays even after a soft reset is *most likely* related to the ESP32 part directly. So not really on the native side written in nf-interpreter neither the C# classes.
Of course, a deeper investigation will be needed!

jefmad, 2021-04-23:
I'm adding to your conclusion that the problem occurs with SSL stream and HttpWebRequest (using SSL), and that the issue appeared after FW 1.4.0.107 on ESP32_WROOM.  I was pointing this : "Improvements in SSL accept handler #62" https://github.com/nanoframework/nanoFramework.System.Net.Http/pull/62 because the change matches the FW, maybe it could be an hint for deeper investigation. Thanks for your help @Ellerbach !

Ellerbach, 2021-04-26:
@jefmad after more tests, you'll have issues as well with the LiLyGo and Ethernet. Same issue. And you don't even need to have SD Card, issue will happen sooner or later.

jefmad, 2021-04-29:
Thanks @Ellerbach . Would you have some time to test and verify the other point in my statement : flashing your device with FW 1.4.0.107 and run the second sample above and ensure that the problem is not here.
This would narrow the moment the problem appeared and confirm where to look and investigate deeper.

Actually my whole home automation project (SD, SSL request, Web server...) is stuck with FW 1.4.0.107 because the bug detailed above is not present, but I'm missing some nice feature like Infra red or others cool things / improvments...

josesimoes, 2021-05-13:
Waiting for migration to next IDF.

jefmad, 2021-07-15:
Since migration of nanoff to cloudsmith, I'm sharing the sources of the Image ESP32_WROOM_32-1.4.0-preview.107 that is the last not having the issue detailed here. 
[ESP32_WROOM_32-1.4.0-preview.107.zip](https://github.com/nanoframework/Home/files/6825805/ESP32_WROOM_32-1.4.0-preview.107.zip)
Use --clrfile with nanoff

josesimoes, 2021-08-11:
@jefmad can you try with the latest fw? The TLS buffer issue has been fixed.

jefmad, 2021-08-12:
@josesimoes Sure, I'll try if it solve the issue, can it wait 2 weeks? I'll give you my feedback

Ellerbach, 2021-08-12:
On my side I can confirm everything is now working perfectly. I did test the scenario with an ESP32 (dev kit, no psram) connected in Wifi using the Azure IoT DPS service with certificate. So it requires a cert2 certificate with private encrypted key, 2 public one, 2 MQTT TLS connection on 2 different servers in a raw. Pure love, all worked like a charm!

jefmad, 2021-08-12:
Hi, after a quick test with an ESP32 WROOM-32 having FW 1.6.5.3 (tested also with FW 1.6.6 preview 530) , running the SSL sample (SecureClient) I have the same issue than detailed above     ++++ System.Net.Security.SslNative::SecureClientInit [IP: 0000] ++++, so I haven't tried to see if it's working with mounting a SD card...
@Ellerbach Can you share your test program, or is this the SSL sample ? Which FW are you running ?

Ellerbach, 2021-08-30:
@jefmad I've been using the basic example. Also I'm using quite intensively TSL with MQTT.
Really all worked. Haven't been testing SSL with the latest framework.
Make sure you're **not** using the BLE version. Just use the normal ESP32_WROOM version.
Happy to do a quick screen sharing, seems like we're both French, can be faster. Ping me on Discord.

jefmad, 2021-08-31:
I confirm that SSL sample and HTTPS sample both work with FW 1.7.0.2, my issue was a matter of PEM certificates to update in the samples. But I cannot test SSL + SD card mounted due to this issue that I'm also reproducing : https://github.com/nanoframework/Home/issues/795
And it seems that System.IO does not have functionality to mount SD card on specific SPI Pins

josesimoes, 2021-09-01:
OK. Thanks for the feedback.

That will be addressed in that other issue.

As it seems to be fixed I'm closing it. Feel free to reopen if needed.

jefmad, 2021-09-01:
Hi @josesimoes , I saw you closed this issue. But to me its status should be blocked (with a reference to https://github.com/nanoframework/Home/issues/795), because the original issue was SSL error when SD card was mounted and nowadays there is no way to mount SD card on ESP 32, so I've not been able to test if the issue is solved. I will be able to test as soon as issue 795 is solved...

josesimoes, 2021-10-13:
It's now possible to mount SDCard on ESP32 and combination of network + TLS has been improved.
Closing this now, feel free to reopen if needed.

closed 2021-10-13 by josesimoes