nanoframework/Home #678: Null terminated string issue

vgolovanov, 2021-01-05:
### Details about Problem

**nanoFramework area:** (C# code )

**VS version:** ( 2019 )

**VS extension version:**

**Target:**ESP32

**Device capabilities output:**

### Description

Null string concatenation not working. Provided code working fine on DotNet Core, but null char is not added to the string on nanoFramework.

### Detailed repro steps so we can see the same problem

```cs
string PropertyValue = "abc";
var oldArray = Encoding.UTF8.GetBytes(PropertyValue);
byte[] newArray = new byte[oldArray.Length + 1];
oldArray.CopyTo(newArray, 0);
string nullTerminatedString = Encoding.UTF8.GetString(newArray, 0, newArray.Length);
var charArray = nullTerminatedString.ToCharArray();

Console.WriteLine(charArray.Length.ToString());

PropertyValue += (char)'\0';
Console.WriteLine(PropertyValue.Length.ToString());

PropertyValue += '\0';
Console.WriteLine(PropertyValue.Length.ToString());

PropertyValue += char.MinValue;
Console.WriteLine(PropertyValue.Length.ToString());
```

### Other suggested things

### Expected behaviour

A clear and concise description of what you expected to happen.

#### Sample Project

### Screenshot

### Additional context

Add any other context about the problem here.

### Make an effort to fix the bug

Attempt to submit a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) that fixes the bug. Include in this PR a test that verifies the fix. If you were not able to fix the bug, a PR that illustrates your partial progress will suffice.

AdrianSoundy, 2021-01-10:
@vgolovanov 
The string implementation in nanoFramework is different to the .Net Core.  In .Net Core the string is internally an array of Char, where a Char is UTF16 and a null is valid.
In nanoFramework the string is a null terminated string of UTF8 characters. If you try to concat a null character you are effectively just added a zero length string so there is no change.
This is by design to reduce the internal size of strings for embedded devices.

AdrianSoundy, 2021-01-10:
This is not a bug so closing issue

closed 2021-01-10 by AdrianSoundy