nanoframework/Home #679: Bug in formatting numbers into a string with D5 format

MateuszKlatecki, 2021-01-07:
### Details about Problem

**Target:** any

**Firmware image version:** newest

**Device capabilities output:**

### Description

The ToString method with the D5 format produces an invalid string for signed numbers.

### Detailed repro steps so we can see the same problem

This program 

```cs
using System;	

public class Program
{
	public static void Main()
	{
		int i = 3;
		Console.WriteLine(i.ToString("D5"));
		
		i = -3;
		Console.WriteLine(i.ToString("D5"));
	}
}
```

Produces output 
```
30000
-3000
```

### Other suggested things

### Expected behaviour

Output should be:
```
00003
-00003
```

### Screenshot

### Additional context

### Make an effort to fix the bug

Attempt to submit a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) that fixes the bug. Include in this PR a test that verifies the fix. If you were not able to fix the bug, a PR that illustrates your partial progress will suffice.

SandorDobos, 2021-02-07:
Are you using an ESP32?
nf_GetFormatString in corlib_native_System_Number.cpp generates this format string: "%-05d" for "D5".
Using this format string in Arduino IDE with the following code results the same bad results as you wrote!

```
// using  Christopher Andrews’ PrintEx library allows use of printf format strings
#include <PrintEx.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  PrintEx serial = Serial;
  serial.printf("[%-05d]\n", -3);
  serial.printf("[%-05d]\n", 3);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

So this looks like not an nf bug, but maybe something we should workaround somehow in nf? :)

BUT: 
Format strings using "-" and "0" aren't valid according to printf's manual page.
Maybe our format string is bad and ESP32's fault is just not taking care well of invalid inputs?

josesimoes, 2021-02-07:
The output in all targets is coming from src\CLR\Helpers\TinyPrintf\printf.c
If anything wrong, it has to be there.

So this is either a bug in the managed code that's requesting a "wrong" format or the implementation of printf.c has a bug somewhere.

josesimoes, 2021-02-07:
Or maybe nf_GetFormatString need to be tweaked to handle with this edge case.

SandorDobos, 2021-02-07:
@josesimoes Are you sure about TinyPrintf? The Arduino code above executed on ESP32 produces the same bad output as @MateuszKlatecki wrote at the beginning. Do you think it is using the same TinyPrintf code somewhere under the hood?

josesimoes, 2021-02-07:
That I don't know... Maybe a coincidence. What I do known is that all our targets are using the code that I've pointed above to output the format on number to string operations. 🙂

Shouldn't be _that_ bad as we haven't received much complaints about this for a long time now. Probably just an edge case needing a tweak.

SandorDobos, 2021-02-07:
I checked TinyPrintf and PrintEx implementations. Both are processing the flags in the same way: 
- if found a '-' : pad = PAD_RIGHT
- if found a '0': pad |= PAD_ZERO
This is against what printf(3) manual page states about '-' and '0' flags:
```
0      The value should be zero padded.  For d, i, o, u, x, X, a, A, e,
              E, f, F, g, and G conversions, the converted value is padded  on
              the  left  with  zeros rather than blanks.  If the 0 and - flags
              both appear, the 0 flag is ignored.  If  a  precision  is  given
              with  a numeric conversion (d, i, o, u, x, and X), the 0 flag is
              ignored.  For other conversions, the behavior is undefined.

       -      The converted value is to be left adjusted on the  field  bound‐
              ary.  (The default is right justification.)  The converted value
              is padded on the right with blanks, rather than on the left with
              blanks or zeros.  A - overrides a 0 if both are given.
```
But I made an other test.
I copied the whole TinyPrintf's printf.c, printf.h file contents with nf_GetFormatString together into [OnlineGDB](https://onlinegdb.com/np8gLUbwo). The output of the `%-05d` format string (associated with `D5` by nf):
- for value 3: "30000"
- for value -3: "-3000"
So it looks the coincidence is working between TinyPrintf and PrintEx and both are bad when interpreting a `-0` combo which is not valid at all :)
Is there any other places than the nf_GetFormatString where such a `-0` combination may be created?
It looks like we should repair at least nf_GetFormatString, but maybe TinyPrintf too to handle `-0` according to ignorance and override mentioned on the above manual.

SandorDobos, 2021-02-07:
@josesimoes I added a "@Jose" comment to the code on (OnlineGDB)[https://onlinegdb.com/np8gLUbwo] which conforms TinyPrintf to the manual. But ng_GetFormatString should be reworked too. Do you want that I try it?

josesimoes, 2021-02-07:
Glad that you've found it! By all means do submit a PR with a fix. 🙂

Also feel free to submit those fixes to the original project although I doubt it will make much difference as it's on full stop for years now.

SandorDobos, 2021-02-07:
Just created an issue for PrintEx and attached the fix there. 
But the original-looking TinyPrintf isn't look like the one included in nf and I found no refs to it's original project :(

josesimoes, 2021-02-07:
> Just created an issue for PrintEx and attached the fix there. 
> But the original-looking TinyPrintf isn't look like the one included in nf and I found no refs to it's original project :(

Strange... I have a vague memory of adding the code from there and including the acknowledgment in the source file headers...
If you found a better replacement for what we have there now, let's not waist time on the current code.et's discuss this on Discord and go for it!

SandorDobos, 2021-02-08:
I checked all implemented standard numeric format strings (to be accurate: the examples on [MS site](https://docs.microsoft.com/en-us/dotnet/standard/base-types/standard-numeric-format-strings) with some trivial variance).
It looks like we have problems not just with D5's above but with these too:

```csharp
(1234).ToString("D6") expected: 001234 found: 123400
(-1234).ToString("D6") expected: -001234 found: -12340
(123.456).ToString("G")) expected: 123.456 found: 123.456000000000003000000000000000
(-123.456).ToString("G")) expected: -123.456 found: -123.456000000000003000000000000000
(123.4546).ToString("G4")) expected: 123.5 found: 123.4546
(-123.4546).ToString("G4")) expected: -123.5 found: -123.4546
((Int32)(-1)).ToString("x")) expected: ffffffff found: FFFFFFFF
(255).ToString("x4")) expected: 00ff found: 00FF
```

Code to test:
```csharp
        public static void Main()
        {
            Debug.WriteLine("Hello from nanoFramework!");

            //AssertToConsole("(123.456).ToString(\"C\"))","",(123.456).ToString("C"));
            //AssertToConsole("(-123.456).ToString(\"C\"))","",(-123.456).ToString("C"));
            //AssertToConsole("(123.456).ToString(\"C3\"))","",(123.456).ToString("C3"));
            //AssertToConsole("(-123.456).ToString(\"C3\"))","",(-123.456).ToString("C3"));
            AssertToConsole("(1234).ToString(\"D\")", "1234", (1234).ToString("D"));
            AssertToConsole("(-1234).ToString(\"D\")", "-1234", (-1234).ToString("D"));
            AssertToConsole("(1234).ToString(\"D6\")", "001234", (1234).ToString("D6"));
            AssertToConsole("(-1234).ToString(\"D6\")", "-001234", (-1234).ToString("D6"));
            //AssertToConsole("(1052.0329112756).ToString(\"E\"))","",(1052.0329112756).ToString("E"));
            //AssertToConsole("(-1052.0329112756).ToString(\"E\"))","",(-1052.0329112756).ToString("E"));
            //AssertToConsole("(1052.0329112756).ToString(\"E2\"))","",(1052.0329112756).ToString("E2"));
            //AssertToConsole("(-1052.0329112756).ToString(\"E2\"))","",(-1052.0329112756).ToString("E2"));
            //AssertToConsole("(1052.0329112756).ToString(\"e\"))","",(1052.0329112756).ToString("e"));
            //AssertToConsole("(-1052.0329112756).ToString(\"e\"))","",(-1052.0329112756).ToString("e"));
            //AssertToConsole("(1052.0329112756).ToString(\"e2\"))","",(1052.0329112756).ToString("e2"));
            //AssertToConsole("(-1052.0329112756).ToString(\"e2\"))","",(-1052.0329112756).ToString("e2"));
            AssertToConsole("(1234.567).ToString(\"F\"))", "1234.57", (1234.567).ToString("F"));
            AssertToConsole("(-1234.567).ToString(\"F\"))", "-1234.57", (-1234.567).ToString("F"));
            AssertToConsole("(1234).ToString(\"F1\"))", "1234.0", (1234).ToString("F1"));
            AssertToConsole("(-1234).ToString(\"F1\"))", "-1234.0", (-1234).ToString("F1"));
            AssertToConsole("(1234.56).ToString(\"F4\"))", "1234.5600", (1234.56).ToString("F4"));
            AssertToConsole("(-1234.56).ToString(\"F4\"))", "-1234.5600", (-1234.56).ToString("F4"));
            AssertToConsole("(123.456).ToString(\"G\"))", "123.456", (123.456).ToString("G"));
            AssertToConsole("(-123.456).ToString(\"G\"))", "-123.456", (-123.456).ToString("G"));
            AssertToConsole("(123.4546).ToString(\"G4\"))", "123.5", (123.4546).ToString("G4"));
            AssertToConsole("(-123.4546).ToString(\"G4\"))", "-123.5", (-123.4546).ToString("G4"));
            AssertToConsole("(1234.567).ToString(\"N\"))", "1,234.57", (1234.567).ToString("N"));
            AssertToConsole("(-1234.567).ToString(\"N\"))", "-1,234.57", (-1234.567).ToString("N"));
            AssertToConsole("(1234).ToString(\"N1\"))", "1,234.0", (1234).ToString("N1"));
            AssertToConsole("(-1234).ToString(\"N1\"))", "-1,234.0", (-1234).ToString("N1"));
            AssertToConsole("(1234.56).ToString(\"N3\"))", "1,234.560", (1234.56).ToString("N3"));
            AssertToConsole("(-1234.56).ToString(\"N3\"))", "-1,234.560", (-1234.56).ToString("N3"));
            //AssertToConsole("(1).ToString(\"P\"))","",(1).ToString("P"));
            //AssertToConsole("(-0.39678).ToString(\"P\"))","",(-0.39678).ToString("P"));
            //AssertToConsole("(123456789.12345678).ToString(\"R\"))","",(123456789.12345678).ToString("R"));
            //AssertToConsole("(-123456789.12345678).ToString(\"R\"))","",(-123456789.12345678).ToString("R"));
            AssertToConsole("(255).ToString(\"X\"))", "FF", (255).ToString("X"));
            AssertToConsole("((Int32)(-1)).ToString(\"x\"))", "ffffffff", ((Int32)(-1)).ToString("x"));
            AssertToConsole("(255).ToString(\"x4\"))", "00ff", (255).ToString("x4"));
            AssertToConsole("((Int32)(-1)).ToString(\"X4\"))", "FFFFFFFF", ((Int32)(-1)).ToString("X4"));

            Debug.WriteLine("--------------------");
        }

        private static void AssertToConsole(string lbl, string expected, string str)
        {
            if (str != expected)
            {
                Debug.WriteLine($"{lbl} expected: {expected} found: {str}");
            }
        }
```

josesimoes, 2021-02-08:
@SandorDobos please take a look at #693.
We better move along and replace the existing code with a decent library and then work on any fixes that might be needed.

SandorDobos, 2021-02-28:
The ToString() reimplementation modifies the deployment item's sizes:
_nanoclr: +3008 byte (compressed: 1300)
mscorlib.dll: -1536 byte
mscorlib.pe: -1200 byte_

josesimoes, 2021-02-28:
> The ToString() reimplementation modifies the deployment item's sizes:
> _nanoclr: +3008 byte (compressed: 1300)
> mscorlib.dll: -1536 byte
> mscorlib.pe: -1200 byte_
> 

Thanks for adding the delta on the assemblies size. Question: have you compare these with RELEASE and MINRELSIZE build?

Care to add this to the PRs too, for sake of completness?

SandorDobos, 2021-02-28:
MinSizeRel nanoclr: +1872 byte (compressed: 1385)
release mscorlib.dll: -1536 byte
release mscorlib.pe: -880 byte

closed 2021-03-02 by josesimoes