nanoframework/Home #682: Setting Generate deployment package results in Error

TitusRie, 2021-01-12:
### Details about Problem

Visual Studio Version 16.8.3
.NET nanoFramework extension v2019.4.2.1
nanoFramework.CoreLibrary Version 1.10.1-preview.9

Target: 
STM32F091 (Nucleo-64)
Firmware 1.6.1-preview.22

### Description

After checking: Settings > Build and Deploy > Generate deployment package 
the deployment will fail with the following error:
[Exception when setting up flash dump: (Value cannot be null.
Parameter name: source
   at System.Linq.Enumerable.First[TSource](IEnumerable`1 source, Func`2 predicate)
   at nanoFramework.Tools.VisualStudio.Extension.DeploymentImageGenerator.<RunPreparationStepsToGenerateDeploymentImageAsync>d__0.MoveNext())]

### Detailed repro steps so we can see the same problem

Connect STM32F091 (Nucleo-64) with USB cable to PC
Update firmware from powershell:
  nanoff --update --target ST_NUCLEO64_F091RC
Create new project Blank Application (nanoFramework)
Select device ST_NUCLEO64_F091RC
Check Settings > Build and Deploy > Generate deployment package 
Start Debugging

### Expected behaviour

Successful deploy and run of application

### Additional context

Output of .NET nanoFramework Extension:
...
2021-01-12 16:59:36Z [NanoDevices: valid device ST_NUCLEO64_F091RC @ COM19 \\?\USB#VID_0483&PID_374B&MI_02#8&13BB54E0&0&0002#]
2021-01-12 16:59:36Z [NanoDevices: Serial device enumeration completed. Found 1 devices]
2021-01-12 16:59:40Z [Check and start debug engine on nanoDevice.]
2021-01-12 16:59:40Z [Connect successful.]
2021-01-12 16:59:40Z [Erase deployment block storage.]
2021-01-12 16:59:40Z [Check device presence...]
2021-01-12 16:59:40Z [Connected to CLR. Pausing execution...]
2021-01-12 16:59:40Z [Erasing sector @ 0x0802A000...]
2021-01-12 16:59:40Z [Rebooting (ClrOnly)]
2021-01-12 16:59:41Z [Reboot command executed successfully]
2021-01-12 16:59:41Z [Waiting for ping event...]
2021-01-12 16:59:41Z [Ping arrived!]
2021-01-12 16:59:41Z [Erase deployment area successful.]
2021-01-12 17:00:04Z [nanoDevice is STM32, checking for flash dump on cache.]
2021-01-12 17:00:04Z [Couldn't find a flash dump for this nanoDevice. Setting up one now. This can take a couple of minutes...]
2021-01-12 17:00:14Z [Exception when setting up flash dump: (Value cannot be null.
Parameter name: source
   at System.Linq.Enumerable.First[TSource](IEnumerable`1 source, Func`2 predicate)
   at nanoFramework.Tools.VisualStudio.Extension.DeploymentImageGenerator.<RunPreparationStepsToGenerateDeploymentImageAsync>d__0.MoveNext())]
2021-01-12 17:00:14Z [Computing deployment blob.]
2021-01-12 17:00:14Z [Deploying assemblies.]
2021-01-12 17:00:21Z [Deploying assemblies. Second attempt.]
2021-01-12 17:00:26Z [Deployment failed.]

TitusRie, 2021-01-12:
Didn't get the template right away, hopefully correct now.
Not sure how to re-open this issue, I'm afraid I don't have the rights for that...

josesimoes, 2021-03-17:
Can't reproduce. Closing.

If this is still an issue for you, please add here a print screen of the Option Dialog for Device Explorer.

closed 2021-03-17 by josesimoes