nanoframework/Home #685: Bugs related to TimeSpan ant Timeout.InfiniteTimeSpan

MateuszKlatecki, 2021-01-19:
### Details about Problem

**Target:** any

**Firmware image version:** newest

Worked before? If so, with which nanoFramework image version: 

**Device capabilities output:** not relevant

### Description

There is few bugs related to TimeSpan. See code below that explain everything:

```cs
using System;
using System.Diagnostics;
using System.Threading;

namespace timeSpanTest
{
    public class Program
    {
        public static void Main()
        {
            TimeSpan minus1ms = TimeSpan.FromMilliseconds(-1);

            //should be true
            Debug.WriteLine($"Timeout.InfiniteTimeSpan == -1 ms : {minus1ms == Timeout.InfiniteTimeSpan }");

            //should sleep forever
            Thread.Sleep(Timeout.InfiniteTimeSpan);

            Debug.WriteLine("Welcome to infinity");

            Timer timer;
            try
            {
                //should create a timer but not start it
                timer = new Timer((_) => { }, null, Timeout.InfiniteTimeSpan, Timeout.InfiniteTimeSpan);
            }
            catch (Exception e)
            {
                //wait, what?
                Debug.WriteLine(e.ToString());
            }

        }
    }
}

```

### Other suggested things

### Expected behaviour

TimeSpan  should work in the same way as in PC.
 

### Additional context

Browsing the history of changes I have the impression that this bug was introduced with this pull https://github.com/nanoframework/nf-interpreter/pull/1724

ddweber, 2021-01-27:
I can confirm that there is something wrong with the Timeout.InfiniteTimespan in the newest version.
I´ve tried to use in a Timer and got following exception:

_++++ Exception System.ArgumentOutOfRangeException - CLR_E_OUT_OF_RANGE (1) ++++
    ++++ Message: 
    ++++ System.Threading.Timer::.ctor [IP: 0000] ++++
    ++++ NewNF.Program::Main [IP: 002c] ++++
Ausnahme ausgelöst: "System.ArgumentOutOfRangeException" in mscorlib.dll_

**Firmware build Info:**
  Date:        Jan 22 2021
  CLR Version: 1.6.1.24

**Assemblies:**
  nanoFramework.Hardware.Esp32, 1.3.2.0
  nanoFramework.Runtime.Events, 1.9.0.0
  mscorlib, 1.10.1.0
  Windows.Devices.Gpio, 1.5.2.0

josesimoes, 2021-01-27:
> I can confirm that there is something wrong with the Timeout.InfiniteTimespan in the newest version.
> I´ve tried to use in a Timer and got following exception:
>
> _++++ Exception [...]

Could you please post the complete line of code throwing the exception?

MateuszKlatecki, 2021-01-28:
@josesimoes see my example code. Passing Timeout.InfiniteTimeSpan to timer (in constructor or change method) throws exception

josesimoes, 2021-01-28:
> @josesimoes see my example code. Passing Timeout.InfiniteTimeSpan to timer (in constructor or change method) throws exception

Understood! (any chance you can debug this and propose a fix?)

SandorDobos, 2021-02-07:
#1724 modified (amongs a lot of code formatting changes :( ) TimeSpan construction to make Timeout.InfiniteTimeSpan's internal value -1 instead of original -1 * TIME_CONVERSION__TICKUNITS.
This makes it's value -00:00:00.0000001 (TotalMilliseconds: -0.00010000, Ticks: -1)
According to .Net docs Timeout.InfiniteTimeSpan's values should look like: -00:00:00.0010000 (TotalMilliseconds:-1, Ticks: -10000)
I think #1724 should be reverted.

josesimoes, 2021-02-07:
Should be fixed now. Please confirm,

closed 2021-02-07 by josesimoes

SandorDobos, 2021-02-07:
Can't confirm yet, building very slow on Docker :)
Your solution looks the same I would make but I didn't understood what edge case the #1724 change added and why we do not simply revert it? The resulting code now looks a bit more complex than the original but has no clear benefit (for me at least :) ).
The only difference is the `(*val == 0)` guard part of the newly (in #1724) introduced expression which isn't looks realistic with the -1 msec value (the caller must have specified positive hour, min, etc. values with -1 msec value).
Just for interest: what was the original problem solved by #1724?
PS: I start a build and hope tomorrow I can confirm the solution :)

josesimoes, 2021-02-07:
Unfortunately the comments on that PR aren't much...detailed and I can't recall exactly what I've stumbled upon back then... My bad. 🙄

I believe it was something with the int64 conversion.

The check could be improved to cover situations on which the call might try to mess up things by putting values for hour, minute, etc and - 1 for milliseconds. Which is a miss use of the intended behavior.

Starting with a check for - 1 value on milliseconds and an else for the rest could improve this. What do you think?

SandorDobos, 2021-02-07:
Looking at `TimeSpan` examples in MS docs I think there are no constraints on creating a `TimeSpan` with mixed positive and negative parameter values at all. Receiving -1 in milliseconds seems valid with any hour, etc. values.

I would just revert the original commit and then look around it's commit date what I was working on, what emails I received, etc.
Maybe something pops up about the int64 conversion problem :)

josesimoes, 2021-02-07:
What would be a use case for positive _others_ and -1 milliseconds??!!

Please recall the context we are in. There a bunch of places that we had to make simplifications and/or choose to have short/few option or even don't have that implementation or overloaded API. So there is nothing wrong to simplify here and make it just reasonable enough.
As long as it's usable and well documented that's OK.

SandorDobos, 2021-02-08:
I don't know the use case which makes this combination valid. But it doesn't mean it has no valid use cases.
I just know that the API allows that and the original code allows that too.

The current code breaks the API and the behaviour we already had.
And the current code is bigger because of extra instructions (guard expressions and value override) :)

The original code converted any value to an internal long. 
If this value is equal to -1 * TIME_CONVERSION__TICKUNITS it should mean the value is infinite.
I think this is simple and compatible and doesn't add complexity.
But we still do not know the reason of current complexity so you should be totally right about the current code's necessarity.

PS.: I suggest to introduce a comment tag, e.g. "APIBREAK: <mandatory short explanation>" for use in all places in nf where because of any reason we decide not to work as expected. 
This would help to distinguish between bugs and "features" too :)

SandorDobos, 2021-02-08:
Meanwhile I can confirm that #1833 fixed the original problem.