nanoframework/Home #687: Support for Raspberry Pi Pico?

theolivenbaum, 2021-01-21:
**nanoFramework area:** Hardware/target board

Just saw on the news that there's a new Raspberry Pi Pico board - and from a quick comparison with other boards here, it seems to be powerful enough to run the nanoframework. 

https://www.raspberrypi.org/documentation/pico/getting-started/

It looks like a really nice board to get started and play around due to the really low cost, and would be a great entry level board for promoting the nanoframework!

Is it already on the roadmap to support it? If not, how would one get started with a custom target for it?

ps: sorry for re-posting this, but my previous #686  issue was closed and I cannot reopen it

networkfusion, 2021-01-21:
@theolivenbaum best join us on our discord channel to discuss there.

josesimoes, 2021-01-21:
@theolivenbaum you _just_ have to follow the template ...

josesimoes, 2021-01-21:
This is being considered. Thanks for pointing it out.

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Once visit this channel for raspberry pi pico projects.

josesimoes, 2021-06-02:
still active

rick-ryerson, 2021-06-11:
Watching to see when this is supported

mguinness, 2022-12-11:
@microhobby has done work to get nanoframework working with NuttX RTOS, see [Running .NET on Raspberry Pi Pico](https://microhobby.com.br/blog/2021/06/24/running-dotnet-on-raspberry-pi-pico/).

arek-w, 2022-12-15:
Solution does not work on raspberry pico W (2022)
I get the message: 
nsh > nsh: nsh_session: readline failed: 11

phantomkingx, 2023-05-20:
....29 months later.

josesimoes, 2023-05-20:
> ....29 months later.

Failing to understand the value that this comment adds to the issue... 🤔 Or to help any ongoing effort to make this happen....

networkfusion, 2023-10-08:
Work is ongoing:

https://github.com/networkfusion/nf-interpreter
https://github.com/TerryFogg/nf-interpreter
https://github.com/burtonrodman/nf-interpreter

pvlst, 2025-01-04:
Any news on this? Or is there any list of what needs to be done in order to get it supported?

Cheers.

josesimoes, 2025-01-06:
pinging @TerryFogg @burtonrodman

burtonrodman, 2025-01-06:
unfortunately _many_ other projects have prevented me from focusing on this (and are still).

if memory serves, i was able to make some minor improvements to the dev container scenario (ie making it easier for new contributors to start).

the furthest i was actually able to get on actual Pico code was reproducing Terry’s work.

TerryFogg, 2025-01-06:
Hi
I have recently returned to development on the Pico series since the release of the pico2.
I have added my code as a community target to my branch as it is easier to integrate any changes from the main as I work.

It should be noted that I perform all of my native development using Visual Studio 2022, I hope someone else can check out development with  VSCode at a later date if they are interested.
I also use CMakepresets.json extensively, and have bypassed the existing Cmake system.

From the previous pull of RP2040 which was custom by me, I have changed it to use the standard pico-sdk build system.

Progress
-----------
I have managed to get ThreadX working on both the Pico ( RP2040) and Pico2 (RP2350), which is the RTOS I am using for this nanoFramework port. The RP2350 was more difficult since I couldn't find any place through my searching that had already done it so I had to spend quite a bit of time getting it going. Although I am not using secure code, the RP2350 boots in secure mode and I had to work through some issues. I have it working, though I haven't dug deep enough to fully understand this aspect ( but its working)

I have GPIO input/outputs working, although I haven't tested out debounce code. The device does have Schmidt trigger feature which might a good thing to implement.
I have SPI ( Graphics) working, on both, but the RP2040 only on small display of 240x135 due to memory limitations.
I have the RP2350 working with a 240x320 display due to it larger internal memory.
I have just purchased a Pimoroni Pico plus 2W which has 16MB Flash and 8MB PSRAM. I am trying to work out how the PSRAM works, I have loaded a standard Pico2 with 4MB flash config and no PSRAM and its works find, just need to understand how to map the PSAM and understand how the boot chooses which chip select for it.

I have I2C communicating to the capacitive touch on my display, so the basics of i2c are working.

After getting the Pimoroni working, I plan to finish off the GPIO,I2C, ADC, PWM.
There is also the wifi/bluetooth, always more work

Terry

josesimoes, 2025-01-06:
WOW @TerryFogg impressive progress!! Thank you for the update.
(to bad you're bypassing the project's CMake, that make it harder to integrate)

TerryFogg, 2025-11-02:
Working on the network when I have time, that's the only barrier at the moment

networkfusion, 2026-03-01:
Support for this via ChibiOS is getting close, so might be easier to implement.