nanoframework/Home #688: nanoFramework.Json hashtable with enumeration not working

opitzh, 2021-01-23:
### Details about Problem

**nanoFramework area:** (C# code | Nuget packages | Community contributions | Documentation )
nanoFramework.Json 2.0.1-preview.53

**VS version:** (2017 | 2019 )
2019

### Description

nanoFramework.Json is not able to create a json string with a hashtable which includes enumerations.

### Detailed repro steps so we can see the same problem

Run below code --> you will get following error message:
++++ Exception System.ArgumentException - 0xfd000000 (1) ++++
++++ Message: 
++++ System.Reflection.MethodBase::Invoke [IP: 0000] ++++
++++ nanoFramework.Json.JsonObjectAttribute::Serialize [IP: 00ac] ++++
++++ nanoFramework.Json.JsonConvert::SerializeObject [IP: 001b] ++++
++++ NF.Test.Program::Main [IP: 0071] ++++

#### Sample Project
var test = new Hashtable
            {
                {DayOfWeek.Monday, "Test"},
                {DayOfWeek.Tuesday, "Test"},
                {DayOfWeek.Wednesday, "Test"},
                {DayOfWeek.Thursday, "Test"},
                {DayOfWeek.Friday, "Test"},
                {DayOfWeek.Saturday, "Test"},
                {DayOfWeek.Sunday, "Test"}
            };

var json = JsonConvert.SerializeObject(test);

networkfusion, 2021-01-23:
It might be worth taking a look at this, as using a hashtable is the old way of serialisation https://github.com/nanoframework/lib-nanoFramework.Json/blob/develop/nanoFramework.Json.Test/Program.cs

opitzh, 2021-01-23:
Thank you very much! :-)
I used an array instead of hashtable and it's working.

closed 2021-01-23 by opitzh