nanoframework/Home #693: Replace tinyprintf implementation

josesimoes, 2021-02-08:
### Details about Problem

**nanoFramework area:**  interpreter

### Description
Need to replace current implementation of printf.
Went back to an old potential candidate from [Kustaa Nyholm](http://www.sparetimelabs.com/tinyprintf/tinyprintf.php).
Reached out to him about the licensing and it's now available as MIT, so we are good to go.

Attached the ZIP file with the sources.
[mit-bsd-lgpl-printf.zip](https://github.com/nanoframework/Home/files/5943709/mit-bsd-lgpl-printf.zip)

### Tasks

- [ ] Replace the current code in `src\CLR\Helpers\TinyPrintf\printf.c` and `src\CLR\Helpers\TinyPrintf\printf.h`.
- [ ] Update copyright headers to match the project ones.
- [ ] Update calls from targets as required.

josesimoes, 2021-02-08:
@SandorDobos would you be interested in taking care of this?

SandorDobos, 2021-02-08:
Kustaa's version lacks some features:
- there is no snprintf
- there is no vsnprintf
- there are no doubles
Should we reimplement these?

josesimoes, 2021-02-08:
> Should we reimplement these?

Considering that:
1. What we have is also needing fixes.
2. We now know the author and can credit it for and reach out for comments and/or contribute back with improvements.
3. Doesn't seem too daunting to add those.

I would say yes. Thoughts?

SandorDobos, 2021-02-08:
Just for test I migrated this one: https://github.com/charlesnicholson/nanoprintf
Very clean implementation. Avoids sprintf and has only sn variants. Much safer :)

But this printf-thing is still one of the cubes composing the Big ToString quest in nf :)
Output still needs tweaking, see #679.

I am going to fix this if I may use nanoprintf or the TinyPrintf code we already have.
I wouldn't dive into Kustaa's implementation if possible.

josesimoes, 2021-02-08:
No problem with using that. The license is OK for the project.
Feel free to proceed.

josesimoes, 2021-02-12:
@SandorDobos just checking if you're tackling this or not. Thanks!

SandorDobos, 2021-02-14:
Yes!

Dobos Sanyi

closed 2021-02-23 by josesimoes