nanoframework/Home #694: Github action not updating dependencies

amgarnier, 2021-02-08:
Please read the following information before posting the issue. Whenever it's clear that the below content was not read, the issue is likely to be closed with "please read the template". Don't take offense at this. It is simply a time management decision. Whenever an issue is raised without reading the template and provide the required information, then often too much time is spent going back and forth to obtain information that is outlined below.

# **Before posting the issue**

If you have a question, need clarification on something, need help on a particular situation or want to start a discussion, do not open an issue here. It is best to ask the question on [Stack Overflow](https://stackoverflow.com/questions/tagged/nanoframework) using the `nanoframework` tag or to start a conversation on one of our [Discord channels](https://discordapp.com/invite/gCyBu8T). Make sure to select the channel that's most appropriate to the context so subject matter experts are most likely to answer promptly.
Issues with questions will most likely be automatically closed.

:exclamation: Remove the content above here and fill out details below. :exclamation:

### Details about Problem

**nanoFramework area:** (C# code | Nuget packages | Community contributions | Documentation )

**VS version:** (2017 | 2019 )

**VS extension version:**

**Target:**

**Device capabilities output:**

### Description

A clear and concise description of what the bug is.

### Detailed repro steps so we can see the same problem

1.

2.

...

### Other suggested things

### Expected behavior

Sample AMQP not resolving references after updating all dependencies

#### Sample Project
https://github.com/nanoframework/Samples/tree/master/samples/AMQP

### Screenshot
[image]

[image]

### Additional context

Discord chat (Zito) that the github action to updated dependencies has stopped.

### Make an effort to fix the bug

amgarnier, 2021-02-08:
Cleaned up the documentation clicked submit too quickly

networkfusion, 2021-02-09:
Added the workflow file.
https://github.com/nanoframework/amqpnetlite/commit/b05ae8cc102d7daa9e9447833eda7e9a1db4ae50

networkfusion, 2021-02-10:
This requires a little more work, but the cause is identified.

josesimoes, 2021-02-12:
Should be fixed now. Check v2.4.2-preview.206.

closed 2021-02-12 by josesimoes

josesimoes, 2021-02-12:
Actually make that 2.4.2-preview.215.