nanoframework/Home #696: Documentation Improvements >> Google Season of Docs

josesimoes, 2021-02-22:
### Details about Problem

**nanoFramework area:** documentation

### Description

1. The project documentation should be audited to clearly identify the improvements necessary to make more usable/discoverable/organized. Following this, a roadmap and list of tasks should exist to help prioritize and distribute work.
1. We should have better getting started guides for the various platforms. The existing ones should be reviewed in content and format. 

### How would we measure success?

- Creation of a documentation roadmap with general approval from the community.
- Increase in number of documentation contributions and new contributors.
- More links to our project from Hackster.io, CodeProject and the vendors that our reference targets are from. 

### What skills would a technical writer need to work on this project?
- Some familiarity with embedded systems, C and/or C#, coding in Visual Studio.
- Some experience doing a docs audit.
- Good organizational skills and hability to perform coordinated tasks (considering that other documentation writers join the project).

### Contact info
Technical writers interested in working on this project please reply to this issue and/or join our Discord community [here](https://discord.gg/gCyBu8T). Kindly include links to your technical writing work or portfolio/résumé/CV.

We've applied to [Google Season of Docs](https://developers.google.com/season-of-docs) to tackle the issues related with our documentation and seek help on improving it.

networkfusion, 2021-04-19:
Close?

josesimoes, 2021-04-19:
Our application was refused by Google. No details where provided just that "There were many high quality applications, so the final selection of organizations was not easy".

josesimoes, 2022-04-29:
Our application was denied. Again. Closing this.

closed 2022-04-29 by josesimoes