nanoframework/Home #697: Http connection to offline url fails with CLR_E_FAIL and results in a total ESP32 hang of all threads for several seconds

skigrinder, 2021-02-24:
### Details about Problem
Attempts to issue an http request (like WebRequest.GetResponse()) to a url that is offline causes all threads on the ESP32 unit to hang for several seconds.

This issue can be reproduced with the following code:
          string testURL  = "http://10.0.6.99/test";     // where 10.0.6.99 is offline
          var httpWebRequest = (HttpWebRequest)WebRequest.Create(testURL);
          using (var httpWebResponse = (HttpWebResponse)httpWebRequest.GetResponse()) {  ...  }

The underlying call to NativeSocket.poll() will fail with CLR_E_FAIL.
This halts processing for the entire ESP32 unit - including threads unrelated to the http request.
The unit hang lasts for several seconds.
A WebException is eventually bubbled up to the caller.

Instead, the WebException should be thrown from the System.Net library in the main thread.
But processing in other threads should not be interrupted

A simple nf project called 'netIssue' is provided to demonstrate this problem
When deployed, netIssue demonstrates a total hang of the ESP32 unit for over 5 seconds.
This would be catastrophic if the ESP32 unit is controlling time-critical hardware. 
For example, a relay connected to a device that burns up if it's not turned off within 700ms of being turned on.

**nanoFramework area:** (C# code | Nuget packages | Community contributions | Documentation )
System.Net Nuget package - specifically the NativeSocket.xxx() functions (i.e. NativeSocket.poll())

**VS version:** (2017 | 2019 )
not relevant - tested with VS 2019 Version 16.8.3

**VS extension version:**
not relevant - tested with nf VS extension version 2019.4.2.6

**Target:**
ESP32 WROOM

**Device capabilities output:**
not relevant

### Description
See above - ### Details about Problem

### Detailed repro steps so we can see the same problem
See above - ### Details about Problem

### Other suggested things

### Expected behaviour
A WebException should be thrown from the System.Net library in the main thread.
But processing in other threads should not be interrupted

#### Sample Project

netIssue.nfproj attached
[netIssue.zip](https://github.com/nanoframework/Home/files/6039311/netIssue.zip)

### Screenshot

### Additional context

Add any other context about the problem here.

### Make an effort to fix the bug

Attempt to submit a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) that fixes the bug. Include in this PR a test that verifies the fix. If you were not able to fix the bug, a PR that illustrates your partial progress will suffice.

skigrinder, 2021-02-24:
Cleaned up instructions.

josesimoes, 2021-03-19:
Can't reproduce....

Fw and class libs updated to versions available today.
Here's the output:

```
03:38:35.900    Main() - Start :
03:38:35.907    wifiSetup() - Attempt to connect to wifi: rede-uai-fai
03:38:45.921    wifiSetup() - Connected to wifi: rede-uai-fai
03:38:45.930    wifiSetup() - ESP32 IP address set to: 0.0.0.0
03:38:46.030    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:46.530    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:47.030    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:47.526    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:48.027    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:48.536    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:49.046    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:49.546    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:50.046    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:50.546    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:51.026    Main() - Issue an HttpWebRequest using a bogus url (i.e. a server that is offline)
03:38:51.032    sendHttpRequest() - sending: http://10.0.6.99/test
03:38:51.064    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:51.566    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:52.066    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:52.566    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:53.066    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:53.566    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:54.066    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:54.576    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:55.076    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:55.576    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:56.076    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:56.576    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:57.076    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:57.576    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:58.076    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:58.576    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:59.076    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:38:59.576    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:39:00.076    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:39:00.576    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:39:01.076    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:39:01.576    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
[... cut ...]
03:39:24.826    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:39:25.326    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:39:25.826    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
03:39:26.326    debugOutput_inSeparateThread() - thread: 0x04 - this output should appear every 500ms without interruption from activity in other threads
```

skigrinder, 2021-03-19:
Hi Jose,
The problem is being reproduced in the ouput below.
Please note the 4.7 sec delay between 
|||||debugOutput_inSeparateThread() |messages.
All processing in all threads (even those not related to the one calling 
sentHttpRequest()) is halted for 4.7 seconds.
Do you agree that this is a big problem for time-critical operations in 
other threads?
Thanks,
    Jim

josesimoes, 2021-03-19:
Will have to dig deeper into this...
Please keep in mind that .NET nanoFramework is **not** an RTOS and time critical operations (really critical!) should be handled in native code. This is not to say that there is nothing wrong there, at this time I can't say, just a heads up.

skigrinder, 2021-03-19:
Thanks Jose.
Understood.
Thanks,
    Jim

skigrinder, 2021-03-20:
Hi Jose',
Another thought...does this issue lend weight to the case for core affinity?
I.e. would threads running on the other core be affected by the freeze 
caused by this issue?
Is it possible to add a 'core' parameter to Thread()?
Thanks,
    Jim

josesimoes, 2021-03-20:
Implementing double core execution, sync and messaging exchange wouldn't be easy...

As for adding a "core" parameter to Thread() wouldn't make it happen because it would require that "another" execution engine would have to be running there.

skigrinder, 2021-03-20:
Hi Jose',
Thanks for clarifying.
Sounds like you've already looked into this.
Are there any lessons to be learned from the Arduino implementation 
(which, I believe, is able to use both cores on the ESP32)?
Thanks,
    Jim

skigrinder, 2021-03-20:
Hi Jose',
Just updated to:
     VS                          16.9.2
     nF extention        2019.5.0.34
     NuGet updates
     nanoff flash         1.6.1-preview.75

Tried to deploy a project to ESP32 but got error:   a2000000

Looks like there may be dependency issues or maybe the NuGets are still 
in a state of flux?
Thanks,
    Jim

skigrinder, 2021-03-21:
Hi Jose',
Just updated to the new NuGets and things are working well again.
Thanks,
    Jim

josesimoes, 2021-03-21:
Well, like in all well, issue gone, or _just_ back to  almost well? 😁

skigrinder, 2021-03-21:
Hi Jose',
I can deploy projects to ESP32 units again but the CLR_E_FAIL issue 
still remains.

Is it possible to change NativeSocket.poll() to return a value of 
'False'  instead of failing (i.e. CLR_E_FAIL) ?
Then a proper SocketException could be thrown from System.Net.Poll() or 
System.Net.Connect() after seeing the 'False' return value.
Please let me know if this makes sense.
Thanks,
    Jim

josesimoes, 2021-03-21:
> Is it possible to change NativeSocket.poll() to return a value of 
> 'False'  instead of failing (i.e. CLR_E_FAIL) ?
> Then a proper SocketException could be thrown from System.Net.Poll() or 
> System.Net.Connect() after seeing the 'False' return value.
> Please let me know if this makes sense.

I'm feeling a strange _deja vu_ feeling about this conversation... 😁

This is unrelated it this issue. Let's discuss this elsewhere.

skigrinder, 2021-03-21:
OK.  I think we're on the same page.

josesimoes, 2021-04-13:
is this still an issue?

skigrinder, 2021-04-14:
Hi Jose',
Thanks for the note and for working on this issue.
Things have greatly improved!
My testing shows that the ESP32 unit hangs (all threads) for a little 
over 3 seconds.
The hang used to be 10-15 seconds.
Using the 'netIssue' project I created for this issue, the user code now 
sees the following.

    03:13:31.962 debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads
    03:13:32.462    debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads
    03:13:32.964    Main() - Issue an HttpWebRequest using a bogus url
    (i.e. a server that is offline)
    03:13:32.970    sendHttpRequest() - sending: http://10.0.6.99/test
         ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL
    (1) ++++
         ++++ Message:
         ++++ System.Net.Sockets.NativeSocket::connect [IP: 0000] ++++
         ++++ System.Net.Sockets.Socket::Connect [IP: 0036] ++++
         ++++ System.Net.HttpWebRequest::EstablishConnection [IP: 018a] ++++
         ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 001a] ++++
         ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++
         ++++ netIssue.netIssue::sendHttpRequest [IP: 001c] ++++
         ++++ netIssue.netIssue::Main [IP: 003f] ++++
         ++++ Exception System.Net.WebException - 0x00000000 (1) ++++
         ++++ Message: connection failed
         ++++ System.Net.HttpWebRequest::EstablishConnection [IP: 019d] ++++
         ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 001a] ++++
         ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++
         ++++ netIssue.netIssue::sendHttpRequest [IP: 001c] ++++
         ++++ netIssue.netIssue::Main [IP: 003f] ++++
    03:13:36.092    debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads
         ++++ Exception System.Net.WebException - 0x00000000 (1) ++++
         ++++ Message: GetResponse() failed
         ++++ System.Net.HttpWebRequest::GetResponse [IP: 00e8] ++++
         ++++ netIssue.netIssue::sendHttpRequest [IP: 001c] ++++
         ++++ netIssue.netIssue::Main [IP: 003f] ++++
    03:13:37.102    debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads
    Exception thrown: 'System.Net.WebException' in System.Net.Http.dll
    03:13:38.102    debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads
    03:13:39.205    debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads
    03:13:39.212    sendHttpRequest() - WebException: GetResponse() failed
    03:13:39.702    debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads
    03:13:40.202    debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads
    03:13:40.702    debugOutput_inSeparateThread() - thread: 0x04 - this
    output should appear every 500ms without interruption from activity
    in other threads

One thought I've had for this issue is the idea of exposing a 
ConnectTimeout property in the HttpWebRequest class.
This would be similar to the existing 'ReadWriteTimeout' property but it 
would be applied to Connect() and subsequently to NativeSocket.connect().
The hope would be that the ConnectTimeout situation would cause a more 
'normal' exception and avoid the CLR_E_FAIL condition.
Maybe this would get rid of the total system hang altogether.

Please let me know if you think this is worthwhile and I'll take a shot 
at it.
Thanks,
     Jim Montgomery
     ***@***.***

josesimoes, 2021-04-16:
The root cause for this may be elsewhere.
Can you add, instead, a thread blinking an LED and have this run standalone, outside VS?
(to give you a heartbeat on the system responsiveness) 

Why? I believe that the hang is being caused by the debugger communication and not the execution engine.
I've been working on a new version of the debugger library that will have the serial communication layer totally rewritten. In the process I've been removing a lot of... inefficiencies... that are currently there. The all communication process is now soo much smooth!

skigrinder, 2021-04-16:
Hi Jose',
I believe you're correct!
If I observe the 'netIssue' test project in the debugger (from VS), I 
see a gap in the heartbeat messages (3-4 seconds).
However, if I use putty to watch the COM port, there's no gap in the 
heartbeat messages!
I believe a blinking LED would show us the same results.
It looks like you're efforts have fixed this issue!!
I'm using putty a lot more lately.  It's a great way to see multiple 
ESP32 units running at the same time.

Here's the output from a 'netIssue' putty session (please also note the 
immediate wifiSetup - which is using the alternate WiFi.Connect() method):

        00:00:00.832 wifiSetup() - Attempt to connect to wifi: layout-2.4
        00:00:01.837 wifiSetup() - Connected to wifi: layout-2.4
        00:00:01.938 wifiSetup() - ESP32 IP address set to: 10.0.6.98
        00:00:01.943 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:02.447 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:02.947 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:03.447 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:03.947 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:04.447 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:04.947 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:05.447 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:05.947 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:06.447 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:06.937 Main() - Issue an HttpWebRequest using a bogus url
        (i.e. a server that is offline)
        00:00:06.942 sendHttpRequest() - sending: http://10.0.6.99/test
             ++++ Exception System.Net.Sockets.SocketException -
        CLR_E_FAIL (1) ++++
             ++++ Message:
             ++++ System.Net.Sockets.NativeSocket::connect [IP: 0000] ++++
             ++++ System.Net.Sockets.Socket::Connect [IP: 0036] ++++
             ++++ System.Net.HttpWebRequest::EstablishConnection [IP:
        018a] ++++
             ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 001a] ++++
             ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++
             ++++ netIssue.netIssue::sendHttpRequest [IP: 001c] ++++
             ++++ netIssue.netIssue::Main [IP: 003f] ++++
             ++++ Exception System.Net.WebException - 0x00000000 (1) ++++
             ++++ Message: connection failed
             ++++ System.Net.HttpWebRequest::EstablishConnection [IP:
        019d] ++++
             ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 001a] ++++
             ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++
             ++++ netIssue.netIssue::sendHttpRequest [IP: 001c] ++++
             ++++ netIssue.netIssue::Main [IP: 003f] ++++
             ++++ Exception System.Net.WebException - 0x00000000 (1) ++++
[... 19 lines cut ...]
        activity in other threads
        00:00:09.487 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:09.987 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:10.487 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:10.987 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:11.487 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:11.987 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads
        00:00:12.487 debugOutput_inSeparateThread() - thread: 0x03 -
        this output should appear every 500ms without interruption from
        activity in other threads

Thanks,
    Jim

josesimoes, 2021-04-16:
@skigrinder good to know!

FYI: we've just released the other week [nanoFramework.Logging](https://github.com/nanoframework/nanoFramework.Logging). It can be very helpful on these situations as it can output as much details as you need without requiring VS to be connected.

Guess this can be close now...

skigrinder, 2021-04-16:
I saw the Logging release and will hopefully give it a try in the coming weeks.
Thanks again for fixing this issue - it was causing me a lot of grief!
I agree we can close #697

closed 2021-04-16 by josesimoes