nanoframework/Home #698: Serialization doesn't work for array in nanoCLR.Win32

Ellerbach, 2021-02-27:
### Details about Problem

**nanoFramework area:** C# code 

**Target:** nanoCLR.Win32.Exe

**Device capabilities output:**

### Description

Problem with serialization, see nanoFramework.Json in the develop-unit-test branch for example.

### Detailed repro steps so we can see the same problem

```csharp
int[] intArray = new[] { 1, 3, 5, 7, 9 };
var result = JsonConvert.SerializeObject(intArray);
```

```
Serialized Array: [1,3,5,7,9]
    ++++ Exception System.NotSupportedException - CLR_E_NOT_SUPPORTED (1) ++++
    ++++ Message:
    ++++ nanoFramework.Json.JsonConvert::Deserialize [IP: 0012] ++++
    ++++ nanoFramework.Json.JsonConvert::DeserializeObject [IP: 0005] ++++
    ++++ nanoFramework.Json.Test.TestJson::Can_serialize_int_array [IP: 0032] ++++
    ++++ System.Reflection.MethodBase::Invoke [IP: 0000] ++++
    ++++ nanoFramework.TestFramework.UnitTestLauncher::RunTest [IP: 004d] ++++
    ++++ nanoFramework.TestFramework.UnitTestLauncher::Main [IP: 007a] ++++
Test failed: Can_serialize_int_array, Exception was thrown: System.NotSupportedException
```

### Expected behaviour

Serialization should work even in nanoCLR.Win32

josesimoes, 2021-02-27:
We only support jagged arrays. Not array of arrays. This is due to the memory constraints and (incredible but expected) computations required.

Could look into it, but I fear it would be "expensive". That's the reason why it wasn't supported in NETMF in the first place.

networkfusion, 2021-02-27:
Strange, because the test passed when I ran it on the device...

josesimoes, 2021-02-28:
> This is just an array of ints, not an array of arrays...
> The test passes when I run on real hardware.

You're correct. In my hurry I overlooked that. This sure looks like a bug.

josesimoes, 2021-03-01:
Just tested this in nanoCLR WIN32 and I can not reproduce:

[image]

Unless I'm looking at the wrong test...

josesimoes, 2021-03-02:
This is caused by the fact that the WIN32 nanoCLR wasn't including the various class libraries. System.Text is one of them.

This is now fixed with version 1.6.1-preview.223 of nanoCLR.

closed 2021-03-02 by josesimoes