nanoframework/Home #702: MDP.Targets failure to build with image added to resource(nanoFramework)

TerryFogg, 2021-03-01:
### Details about Problem

**nanoFramework area:** ( Visual Studio extension | Other tools )

### Description

To Reproduce

Create new nanoFramework blank application
Add resource file (nanoFramework)
Open resource file and select images
Drag image from windows folder to add image to resource file
Save
Build
Error
.NET nanoFramework GenerateNanoResourceTask error: A generic error occurred in GDI+. NFApp2 C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets 709

### Other suggested things

### Expected behaviour

A clear and concise description of what you expected to happen.

Successful Build

### Additional context

Hoped to use a bitmap for graphics

### Make an effort to fix the bug

josesimoes, 2021-03-01:
@TerryFogg seems that you've left content from the instructions and that was causing the issue to be close because of wrong formatting.

josesimoes, 2021-03-03:
Just followed the above on VS 2019 16.8.5 and extension 2019.5.01 without any problems.
Image added as a PNG.

TerryFogg, 2021-03-03:
Mine was a jpg, I will check again today

josesimoes, 2021-03-17:
@TerryFogg can you please check this and update or close the issue? 😉

TerryFogg, 2021-03-17:
_Note 1: A general problem , on my version (using dark theme) there is an update display issue of the Resource1.resx when displayed.  After closing and reopening the designer, it displays black ( dark theme background). To force it to display properly I had to load Program.cs and then go back and it displayed properly._

_Note 2: All these tests are at the design/build stage and I know there are additional problems to debugged in the native code to support tests. ( current samples use an initialized byte array to get images into the program)_
**1.	 PNG file test**
Drag PNG to design surface and build – OK
**2.	Static GIF file test ( source file GIF01Normal.gif from .net MicroFramework sources)**
Drag GIF to design surface and build – OK
**3.	BMP file test ( source file DefaultButtonUp.gif from .net MicroFramework sources)**
Build Failed
.NET nanoFramework GenerateNanoResourceTask error: Exception of type 'System.OutOfMemoryException' was thrown.
**4.	JPG file test ( source file Tom_and_Jerry.jpg from .net MicroFramework sources)**
Build Failed
.NET nanoFramework GenerateNanoResourceTask error: A generic error occurred in GDI+.

josesimoes, 2021-03-17:
Could you please attach here those two that are failing for you?

closed 2021-03-23 by josesimoes