nanoframework/Home #704: Unit Test migration and elements from .NET Microframework

Ellerbach, 2021-03-02:
### Details about Problem

Now we have a great Unit Framework, that work is almost finished to migrate and adapt tests from .NET Microframework for mscorlib, this task to prepare the work for adding Unit Test into other libraries and inventory of what can be reused with adaptation for nanoFramework.

### Description

This is the work to be done. Please volunteer to work on any of those! It's a great way to learn about the framework, what's inside.

Core path: Test/Platform/CLR

| Path | Target lib | Note |
| --- | --- | --- |
| Microsoft.SPOT.Graphics | N/A | Can reuse the graphics in this file already in byte array |
| Microsoft.SPOT.Hardware/EmulateHardware | Potentially nanoff/tools | Some code and ideas to reuse for the nanoff and other hardware related tests. That said, as most are happening in .NET 4.x, .NET Core/5.0, this can't be reused directly like that. Some of the TestPower and maybe TestWatchdog can be reused and adapted. |
| Microsoft.SPOT.Hardware//I2C | N/A | Has to be rewritten for real hardware test with a subset of real hardware. Here an EPROM is used. |
| Microsoft.SPOT.Hardware/LargeBufferTests | lib-CoreLibrary | Paradigm changed. Still need to add tests on buffers. |
| Microsoft.SPOT.Hardware/UsbClient | Potentially nanoff/tools  | Same as for the Emulated hardware, some tests will have to be added but more regarding  nanoff or the debug sessions. |
| Microsoft.SPOT.Interop | We do have most of the Microsoft.Devices and Microsoft.System libraries using this mechanism. Those are tests per say and it doesn't require specific tests. If that breaks, all the libraries will break. We can consider them all as tests in this case. |
| Microsoft.SPOT.Net | lib-nanoFramework.System.Net, lib-Windows.Devices.WiFi and specific hardware ones | IP, DNS and other tests that can be easily adapted. |
| Microsoft.SPOT.Security/ClientServer | lib-nanoFramework.System.Net.Http | Elements to migrate and reuse. |
| Microsoft.SPOT.Security/SslStream | lib-nanoFramework.System.Net.Http | Elements to migrate and reuse. |
| Microsoft.SPOT.Time/TimeServices | lib-nanoFramework.Networking.Sntp | Elements to migrate and reuse. |
| Microsoft.SPOT.Time/TinyCore/* | lib-nanoFramework.Graphics | Elements to migrate, adapt and reuse. |
| mscorlib/* | mscorlib | Work done. Exception /collections to move to lib-nanoFramework.System.Collections. |
| system/Http | lib-nanoFramework.System.Net.Http | To almost fully reuse. |
| system/IO | lib-Windows.Storage.Streams | To almost fully reuse with some adaptation. |
| system/IO | lib-nanoFramework.System.Net | To almost fully reuse with some adaptation. |
| system/Security | TBD | TBD |
| system/socketTests | lib-nanoFramework.System.Net | To almost fully reuse with some adaptation. |
| system/SPOTXmlTests | N/A | N/A |
| system/Text | lib-nanoFramework.System.Text | To almost fully reuse. |
| system/xmlTests| N/A | N/A |

Code path: Test/Platform/Performance

| Path | Target lib | Note |
| --- | --- | --- |
| ProfilerTests/* | Depending on the tests: Collections, String, Thread | Almost al can be reused and adapted, great for performance stress tests. |
| Stress/Loopback | lib-nanoFramework.System.Net, Thread | A lot can be reused and adapted, great for performance stress tests. |
| String | lib-nanoFramework.System.Text, mscorlib | To almost fully reuse with some adaptation. |

edleno2, 2021-05-02:
@Ellerbach - are all of these new unit tests?  I'm looking for a "good first issue" and this looks like fresh unit tests and basically no potential for needing to change "real" core code.  Is my assumption right?

Ellerbach, 2021-05-04:
> are all of these new unit tests?

The core idea is to reuse some of the tests written for Microframework: https://github.com/NETMF/netmf-interpreter/tree/dev/Test
There are existing ones, some great ones, some which need to be adjusted. The test framework used is different. So you'll have to do extended replace for logs to Debug and the function definition, adding the Assert where it makes sense.
The table here explains what you can reuse. Some of the existing nanoFramework classes already have migrated and newly written unit tests. Recommendation is to start by migrating existing ones then adding new ones to cover new scenarios.
I hope that makes sense :-)

edleno2, 2021-05-04:
> The table here explains what you can reuse. Some of the existing nanoFramework classes already have migrated and newly written unit tests. Recommendation is to start by migrating existing ones then adding new ones to cover new scenarios.
> I hope that makes sense :-)

Ok, I found the netmf tests and I think I understand what this is about.  I'm going to first pick up the issue on some of the failing unit tests - I think by changing existing tests first I'll get an understanding about how the unit test framework works, and then I can go on to writing new tests under this issue.  I'll add myself to the other issue and come back to this one later.

rschiefer, 2022-10-18:
I checked several of these and couldn't find any that weren't migrated over.  Any that we know still need to be added?

Ellerbach, 2022-10-19:
> I checked several of these and couldn't find any that weren't migrated over. Any that we know still need to be added?

As far as I remember, the networking one hasn't been migrated as they all require hardware tests. Like HTTP. So that maybe a good one to move forward. Yes, it will require to have devices connected but we're thinking of real hardware tests time to time to check all works fine. Basically connected to a RPI with the Azure agent on them to run them.

josesimoes, 2022-10-19:
Good point! Maybe it's time that we close this one.
When those hardware and network tests are possible we'll revisit the old test and see what we can salvage.

Are we in agreement? 😃

Ellerbach, 2022-10-19:
We can move them as well as unit test, they'll be there and we can use the pattern already in place for SPI, Serial and few others where they are skipped if hardware does not support.

Ellerbach, 2023-04-18:
I4ll close this issue as we've been reusing all what was needed and still reuse when we move things forward. There's not really any "normal" test to migrat anymore!

closed 2023-04-18 by Ellerbach