nanoframework/Home #705: The EnableWakeupByPin not waking up ESP32

AdrianSoundy, 2021-03-04:
### Description

The function to wake up by a single pin no longer seems to work (EnableWakeupByPin) in the class libray nanoFramework.Hardware.Esp32. Althought the EnableWakeupByMultiPins still seems to work.

The ESP32 never wakes up.

### Expected behaviour
The ESP32 wakes up when pin level changed to wake up level.

#### Sample Project
Using wake sample project

hbraasch, 2021-03-04:
The issue is slightly more involved. Even when wakeup works using EnableWakeupByMultiPins, as soon another pin gets added to the program and configured (as simple as adding it as another input) the EnableWakeupByMultiPins also stops working. It then appears as if the  MCU resets as soon as sleep is entered.
The sample program demonstrating this is attached below.

[DeepSleepTesterAdditional.txt](https://github.com/nanoframework/Home/files/6082884/DeepSleepTesterAdditional.txt)

hbraasch, 2021-03-05:
Repeated the test using the ESP-IDF. It works as its suppose to.

[image]

So the problem unfortunately lies with nanoFramework.

closed 2021-03-08 by josesimoes