nanoframework/Home #707: NFUnitTestSystemLib.SystemType1_GetType_Test fails

josesimoes, 2021-03-04:
### Details about Problem

**nanoFramework area:** (mscorlib)

### Description

Test is failing. Need to fix code in mscorlib.

### Screenshot

[image]

SandorDobos, 2021-03-08:
this test is green for me.
my nanoframework.TestFramework is at b0f34e49da66edfc066793005e7c45217266fed4.

Ellerbach, 2021-03-09:
> this test is green for me.

The failing ones are the ValueArrayTypes. In short GetType("System.Int32[]") and any other with array is not working
[image]

josesimoes, 2021-06-03:
Fixed!

closed 2021-06-03 by josesimoes