nanoframework/Home #709: Several UnitTestParseTests.Parse are failing

josesimoes, 2021-03-04:
### Details about Problem

**nanoFramework area:** (mscorlib )

### Detailed repro steps so we can see the same problem

Several Parse calls are failing. The all code section needs to be reviewed. Evaluate if this is better fixed at managed or native end. OK to move to native if more efficient.

### Screenshot

[image]

edleno2, 2021-05-25:
@josesimoes - so in one of the cases the problem is that the native code isn't generating an exception - so something like Byte.Parse("") gets a FormatException in .net Core.  So for practice I added the code to nanoCLR and CoreLibrary to add System.FormatException and I got it all to work.  Wanted to understand really how the checksums and c#->C++ interop work.  Is it Ok if I check that in?  It does mean new versions of firmware go to with a new version of CoreLibrary that wasn't really necessary.  But maybe better for me to learn how on something small like this - but, may break the build if I don't get everything right.  I think I have to manually change the version in corlib_native.cpp to the next version.  Do I also bump the version in version.json in CoreLibrary - the preview number will bump anyhow.  Will the build process update the message in Nuget about the required framework version or is that a manual process?

josesimoes, 2021-05-25:
> @josesimoes - so in one of the cases the problem is that the native code isn't generating an exception - so something like Byte.Parse("") gets a FormatException in .net Core. 

Understood and that makes sense.

> So for practice I added the code to nanoCLR and CoreLibrary to add System.FormatException and I got it all to work. Wanted to understand really how the checksums and c#->C++ interop work. Is it Ok [...]

If I understood this correctly, it's a matter of throwing that exception right from the C# code. No need to change anything at the native end.

> But maybe better for me to learn how on something small like this - but, may break the build if I don't get everything right. I think I have to manually change the version in corlib_native.cpp to [...]

That's basically it. 1) you make the changes 2) bump assembly native version (if anything is changing in the API) 3) generate the stubs 4) update the declaration of the CorLib in native 5) adjust whatever is required there.
All the rest is automated.

josesimoes, 2021-06-03:
Fixed!

closed 2021-06-03 by josesimoes