nanoframework/Home #712: Can't use templates in VS 2019 v16.9

josesimoes, 2021-03-05:
### Details about Problem

**nanoFramework area:** ( Visual Studio extension )

**VS version:** 16.9

### Description

Starting with v16.9 project templates don't work anymore.
Existing solutions keep working just fine.

### Additional context

Seems to be an issue related with NuGet.
Reported to VS Developer Community  **>>[here](https://developercommunity.visualstudio.com/t/nanoframework-templates-can-no-longer-be-created/1358051)<<**

# ⚠️ **PLEASE UPVOTE** THE ISSUE IN VS Developer Community to speed up resolution ⚠️

rrelyea, 2021-03-06:
debugged this...your project system says it has cps capabilities and packagereference capabilities.
this causes us to try PackageReferences, but that is different than 16.8.

Since you need packages.config, like 16.8, you should not advertise packagereference capability.

NFProjectSystem.targets -- line 20 - contains PackageReferences.
When i remove that, things seem to work fine with 16.9

josesimoes, 2021-03-06:
@rrelyea indeed that fixes it! 💯 
Thank you so much for looking at it in so short notice. And offering a fix, of course. 😉

closed 2021-03-06 by josesimoes