nanoframework/Home #719: Device Explorer not seeing STM32F746 Nucleo

dazzalogan, 2021-03-12:
### Details about Problem

Hi,

I have a Nucleo-F746ZG board but VS2019 Device Explorer is not seeing it connected.
Both USB connections are made to the board.

Package manager console output:

PM> nanoff --target ST_NUCLEO144_F746ZG --update -v diag
nanoFramework Firmware Flasher v1.15.0+96f858f13a
Copyright (C) 2019 nanoFramework project contributors

Trying to find ST_NUCLEO144_F746ZG in developement repository...
Trying to find ST_NUCLEO144_F746ZG in community targets repository...OK
Downloading firmware package...OK
Download location is C:\Users\Darren.Logan\AppData\Local\Temp\b307315b-7dcf-4c55-97e7-79cedd82957a
Updating to 1.6.929.0
Extracting ST_NUCLEO144_F746ZG-1.6.929.0.zip...
Error E1000: No DFU device found. Make sure it's connected and has booted in DFU mode
PM>

Apparently the issue is related to DFU mode, but I'm out of my depth with this - so please help.

Thank you

Regards

**nanoFramework area:** ( Visual Studio extension | Other tools )

**VS version:**

**VS extension version:**

**Target:**

**Firmware image version:**

**Device capabilities output:**

### Description

A clear and concise description of what the bug is.

### Detailed repro steps so we can see the same problem

1.

2.

...

### Other suggested things

### Expected behaviour

A clear and concise description of what you expected to happen.

### Screenshot

### Additional context

Add any other context about the problem here.

### Make an effort to fix the bug

Attempt to submit a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) that fixes the bug. Include in this PR a test that verifies the fix. If you were not able to fix the bug, a PR that illustrates your partial progress will suffice.

networkfusion, 2021-03-13:
As a work around for this (since the board does not seem to be able to handle DFU packages), download the firmware from https://bintray.com/nfbot/nanoframework-images-community-targets/ST_NUCLEO144_F746ZG/1.6.930.0#files and use ST-Link utility to flash the HEX files for nanoBooter and nanoClr

networkfusion, 2021-03-13:
Possible duplicate of #683

piwi1263, 2021-03-13:
@dazzalogan @josesimoes are you using the lates of nanoff since I've got v1.16.0-preview.3+4e0b0c3891. Plus do put your board in DFU mode and try nanoff --listdfu the dfuid should be shown.

closed 2021-03-13 by networkfusion