nanoframework/Home #725: Unhandled exception in System.Device.Gpio (GpioControllerValueChanged / RegisterCallbackForPinValueChangedEvent)

Karlfried, 2021-03-30:
Unhandled exception in System.Device.Gpio (GpioControllerValueChanged / RegisterCallbackForPinValueChangedEvent).

I tried to trigger an event via an input change [pin 1-0 edge].
That then turned into an unhandled exception.
I tested this on 2 Borads (ESP32_WROMM_32 & ST_NUCLEO64_F091RC) with slightly different results.

Microsoft 2019 / Version 4.8.04084
.Net nanoFramework VS2019 Extension / Version 2019.5.0.40
nanoFramework.CoreLibrary / Version 1.10.3-preview.7
nanoFramework.Runtime.Events / Version 1.9.0-preview.16
nanoFramework.System.Device.Gpio / Version 1.0.0-preview.31
System.Device.Gpio  v100.1.0.4

**VS2019-Output:** ESP32_WROMM_32 [Board-Version: ESP32_WROOM_32-1.6.1-preview.76]**

`The nanoDevice runtime is loading the application assemblies and starting execution.
"nanoFramework.Tools.VS2019.Extension.dll" (Verwaltet): "E:\Temp\_NanoFramework\GpioControllerValueChanged\bin\Debug\GpioControllerValueChanged.exe" geladen, Symbole geladen.
"nanoFramework.Tools.VS2019.Extension.dll" (Verwaltet): "E:\Temp\_NanoFramework\GpioControllerValueChanged\packages\nanoFramework.Runtime.Events.1.9.0-preview.16\lib\nanoFramework.Runtime.Events.dll" wurde geladen, das Laden von Symbolen wurde übersprungen. Das Modul ist optimiert, und die Debugoption "Nur eigenen Code" ist aktiviert.
"nanoFramework.Tools.VS2019.Extension.dll" (Verwaltet): "E:\Temp\_NanoFramework\GpioControllerValueChanged\packages\nanoFramework.System.Device.Gpio.1.0.0-preview.31\lib\System.Device.Gpio.dll" wurde geladen, das Laden von Symbolen wurde übersprungen. Das Modul ist optimiert, und die Debugoption "Nur eigenen Code" ist aktiviert.
Der Thread '<Kein Name>' (0x2) hat mit Code 0 (0x0) geendet.
Time-Ticker! [1]
Time-Ticker! [2]
Time-Ticker! [3]
    ++++ Exception System.ArgumentException - 0xfd000000 (3) ++++
    ++++ Message: 
    ++++ System.Enum::HasFlag [IP: 0000] ++++
    ++++ System.Device.Gpio.GpioController::GpioControllerValueChanged [IP: 0045] ++++
    ++++ System.Device.Gpio.GpioPin::OnPinChangedInternal [IP: 002d] ++++
    ++++ System.Device.Gpio.GpioPinEventListener::OnEvent [IP: 002a] ++++
    ++++ nanoFramework.Runtime.Events.EventSink::ProcessEvent [IP: 0023] ++++
    ++++ nanoFramework.Runtime.Events.EventSink::EventDispatchCallback [IP: 0014] ++++
Ein Ausnahmefehler des Typs "System.ArgumentException" ist in mscorlib.dll aufgetreten.
`

**VS2019-Output: ST_NUCLEO64_F091RC [Board-Version: ST_NUCLEO64_F091RC-1.6.2-preview.17]**

`The nanoDevice runtime is loading the application assemblies and starting execution.
Time-Ticker! [1]
Time-Ticker! [1]
Time-Ticker! [2]
Time-Ticker! [3]
Time-Ticker! [4]
Time-Ticker! [5]
 Uncaught exception 
Time-Ticker! [6]
Failed allocation for 17 blocks, 204 bytes

 Uncaught exception 
Time-Ticker! [7]
Time-Ticker! [8]
 Uncaught exception 
 Uncaught exception 
 Uncaught exception 
Time-Ticker! [9]
Time-Ticker! [10]
Time-Ticker! [11]
`

Sample Project:

`using System;
using System.Device.Gpio;
using System.Diagnostics;
using System.Threading;

namespace GpioControllerValueChanged
{
    public class Program
    {
        static GpioController _gpioController;
        static int BlueLedPinNumber;
        static int UserButtonPinNumber;
        static int TickerCounter;
        private static int UserButtonCounter;

        public static void Main()
        {
            _gpioController = new GpioController();

            // ESP32 (LOLIN32)
            BlueLedPinNumber = 5;
            UserButtonPinNumber = 13;
            

            // STM32_NUCLEO64_F091RC
            //BlueLedPinNumber = PinNumber('A', 5); // LED green
            //UserButtonPinNumber = PinNumber('A', 0); // A0

            /////////////////////
            // setup blue LED //
            /////////////////////
            _gpioController.OpenPin(BlueLedPinNumber, PinMode.Output);

            ///////////////////////
            // setup user button //
            ///////////////////////
            
            // As input
            _gpioController.OpenPin(UserButtonPinNumber, PinMode.InputPullUp);

            // Register event
            _gpioController.RegisterCallbackForPinValueChangedEvent(UserButtonPinNumber, PinEventTypes.Falling | PinEventTypes.Rising, UserButton_ValueChanged);

            while (true)
            {
                _gpioController.Write(BlueLedPinNumber, PinValue.High);
                Thread.Sleep(2000);
                _gpioController.Write(BlueLedPinNumber, PinValue.Low);
                Thread.Sleep(2000);

                TickerCounter++;
                Debug.WriteLine("Time-Ticker! [" + TickerCounter + "]");
            }

            Thread.Sleep(Timeout.Infinite);
        }

        /// <summary>
        /// User button Event
        /// </summary>
        /// <param name="sender"></param>
        /// <param name="e"></param>
        private static void UserButton_ValueChanged(object sender, PinValueChangedEventArgs e)
        {
            // read Gpio pin value from event
            UserButtonCounter++;
            Debug.WriteLine("USER BUTTON event[" + UserButtonCounter + "] : " + e.ChangeType.ToString());

            if (e.ChangeType == PinEventTypes.Rising)
            {
                _gpioController.Write(BlueLedPinNumber, PinValue.High);
            }
            else
            {
                _gpioController.Write(BlueLedPinNumber, PinValue.Low);
            }
        }

        static int PinNumber(char port, byte pin)
        {
            if (port < 'A' || port > 'J')
                throw new ArgumentException();

            return ((port - 'A') * 16) + pin;
        }
    }
}`

Ellerbach, 2021-03-31:
Can you try with a lock object when writing on the pins? The GpioController is not thread safe. You have to take care of this part. That's maybe the issue.

Karlfried, 2021-04-01:
I removed the access to 'GpioController' in the event function.
With a 1-0 edge on the pin, an unhandled exception still occurs on both boards. 

New event function:

```
private` static void UserButton_ValueChanged(object sender, PinValueChangedEventArgs e)
        {                
               UserButtonCounter++;
        }
````

SandorDobos, 2021-04-12:
Not a Gpio problem, see #730.

josesimoes, 2021-04-12:
Fixed through nanoframework/nf-interpreter#1882. If this is not the case, please reopen the issue.

closed 2021-04-12 by josesimoes