nanoframework/Home #727: nanoFramework VS2019 Extension - Device Explorer: No serial NanoDevices found! [USB to serial converter chip CH340]

Karlfried, 2021-04-03:
### Details about Problem

nanoFramework VS2019 Extension - Device Explorer: No serial NanoDevices found! [USB to serial converter chip CH340]

**nanoFramework area:** Visual Studio extension

**VS version:** 2019 V 16.9.1

**VS extension version:** 2019.5.0.40 

**Target:** ESP32_WROOM_32

**Firmware image version:** ESP32_WROOM_32-1.6.1-preview.75

Windows 10 - USB to serial converter chip CH340 - Driver Version: 3.5.2019.1 [2019/01/30]
Microsoft VS2019 / Version 16.9.1
.Net nanoFramework VS2019 Extension / Version 2019.5.0.40 
Microsoft VS2017 / Version 15.9.27
.Net nanoFramework VS2017 Extension / Version 2017.1.8.16

### Description

I have an ESPDUNO-32 board. This board uses a CH340 chip for the USB to serial converter.
Until a few weeks ago, the board was recognized in the Device Explorer in VS2017 and also in VS2019.
Since the update of the 'nanoFramework VS2019 Extension', the board is no longer recognized in the Device Explorer of VS2019.
In VS2017 it works perfectly.

**VS2017-Output: --> 1 devices found**
```
2021-04-01 16:24:02Z [NanoDevices: new device arrival \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3# ]
2021-04-01 16:24:02Z [NanoDevices: candidate nano device \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3#]
2021-04-01 16:24:02Z [NanoDevices: checking device  \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3# @ 921600]
2021-04-01 16:24:03Z [NanoDevices: checking device  \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3# @ 460800]
2021-04-01 16:24:04Z [NanoDevices: checking device  \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3# @ 115200]
2021-04-01 16:24:06Z [NanoDevices: checking device  \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3# *** 2nd attempt ***]
2021-04-01 16:24:06Z [NanoDevices: checking device  \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3# @ 921600]
2021-04-01 16:24:06Z [NanoDevices: valid device ESP32_WROOM_32 @ COM3 \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3#]
2021-04-01 16:24:06Z [NanoDevices: Serial device enumeration completed. Found 1 devices]
```

**VS2019-Output: --> 0 devices found**
```
2021-04-01 16:28:54Z [NanoDevices: new device arrival \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3# ]
2021-04-01 16:28:54Z [NanoDevices: candidate nano device \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3#]
2021-04-01 16:28:54Z [NanoDevices: dropping black listed device \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2021-04-01 16:28:56Z [NanoDevices: checking device  \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3# *** 2nd attempt ***]
2021-04-01 16:28:56Z [NanoDevices: dropping black listed device \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3#{86e0d1e0-8089-11d0-9ce4-08003e301f73}]
2021-04-01 16:28:56Z [NanoDevices: quitting device \\?\USB#VID_1A86&PID_7523#6&30D39121&0&3#]
2021-04-01 16:28:56Z [NanoDevices: Serial device enumeration completed. Found 0 devices]
```

josesimoes, 2021-04-03:
@Karlfried there was really nothing changed on the communication layer that was changed/updated for a long time...
The changes in the extension should be unrelated.

Having said that, some follow up questions:
1) Are you using this in the same machine?
2) If, yes, are you disabling the device watchers in one of the VS instances? 

The detection process tries to communicate with any COM port waiting for a valid answer.
If the detection is ongoing on the _other_ VS instance, 99% sure that it will fail on because it won't be able to communicate with the COM port as it's being used.

I notice on the log above that on VS 2019, it's not even trying to communicate with `USB#VID_1A86&PID_7523#6&30D39121&0&3#{86e0d1e0-8089-11d0-9ce4-08003e301f73}` because it's on the black list.
You must have add it there, therefore it can't possibly be found.

Karlfried, 2021-04-03:
@josesimoes yes, it is always the same PC and the same board that I used for this test.
There is also only 1 VS open at a time.

But you are right, the COM port was in the black list.
It is a mystery to me how that could happen.
I didn't know the black list before.
Now it works perfectly again.

**Many thanks and sorry about that.**

closed 2021-04-03 by Karlfried