nanoframework/Home #729: Nanoframework firmware image with Ethernet support for Olimex ESP32-POE board.

Shraddha246, 2021-04-09:
Nanoframework firmware image with Ethernet support for Olimex ESP32-POE board.
We have tried to update NFinterpreter and flashed it on Olimex ESP32-POE board.
We got IP address but we are getting socket exception while connecting to MQTT service. Is there any plan to create separate firamware for ESP32 with Ethernet support. Please let me know.  Thanks.

closed 2021-04-09 by nfbot

toddjohnson37, 2022-06-27:
Where you ever able to use the Olimex ESP32-POE with NanoFramework. I was able to flash it with nanoff and it reports back with ESP32 (ESP32-D0WD-V3 (Revision 3) with the features of WiFi, BT, Dual Core, 240MHz, but does not list Ethernet. I can run a basic program, but can not get the ethernet port to work?  Thanks, Todd

Shraddha246, 2022-10-11:
Hi,
      There is no ready to deploy nanoframework firmware for ESP32 with Ethernet support.
We need to build our own firmware and flash it.
We need to make changes in following files in repository:
nf-interpreter<https://github.com/nanoframework/nf-interpreter>/targets<https://github.com/nanoframework/nf-interpreter/tree/main/targets>/ESP32<https://github.com/nanoframework/nf-interpreter/tree/main/targets/ESP32>/_include<https://github.com/nanoframework/nf-interpreter/tree/main/targets/ESP32/_include>/esp32_idf.h
//Uncomment to support Ethernet
//#define ESP32_ETHERNET_SUPPORT 1

nf-interpreter<https://github.com/nanoframework/nf-interpreter>/targets<https://github.com/nanoframework/nf-interpreter/tree/main/targets>/ESP32<https://github.com/nanoframework/nf-interpreter/tree/main/targets/ESP32>/_Network<https://github.com/nanoframework/nf-interpreter/tree/main/targets/ESP32/_Network>/NF_ESP32_Ethernet.cpp
#define ETH_PHY_RST_GPIO 12
#define ETH_RMII_CLK_OUT_GPIO 17

To build and flash firmware refer following link:
How to Build, Flash and Debug the ESP32 nanoCLR on Windows using Visual Code | nanoFramework Documentation<https://docs.nanoframework.net/content/building/build-esp32.html>
How to Build, Flash and Debug the ESP32 nanoCLR on Windows using Visual Code - nanoFramework<https://docs.nanoframework.net/content/building/build-esp32.html>
How to Build, Flash and Debug the ESP32 nanoCLR on Windows using Visual Code. ⚠️ NOTE about the need to build .NET nanoFramework firmware ⚠️. You only need to build it if you plan to debug the CLR, interpreter, execution engine, drivers, add new targets or add new features at native level.
docs.nanoframework.net

Regards,
Shraddha

josesimoes, 2022-10-11:
@Shraddha246 we have ESP32_OLIMEX image ready to use. Recently there have been a few more CMake presets added to allow building it for other ethernet enabled targets.

If this is not working for you, I suggest that you join our Discord [here](https://discordapp.com/invite/gCyBu8T) and start a conversation about this.

muenchris, 2024-05-11:
> @Shraddha246 we have ESP32_OLIMEX image ready to use. Recently there have been a few more CMake presets added to allow building it for other ethernet enabled targets.
>
> If this is not working for [...]

The ESP32_OLEMEX 1.8.1.93 does not show the Ethernet port during SetupAdnConnectNetwork. Is the latest image broken or does the ethernet has to be enabled somehow? The device Explorer does not show an ethernet selection in the network settings

belem2050, 2025-01-22:
By enabling staticipV4, it worked for me:

`IPConfiguration IPConfiguration = new IPConfiguration("192.168.31.147", "255.255.255.0", "192.168.31.1", new string[] {"8.8.8.8"});
// 0 was the index of the interface I wanted to connect to
 NetworkInterface.GetAllNetworkInterfaces()[0].EnableStaticIPv4(IPConfiguration.IPAddress, IPConfiguration.IPSubnetMask, IPConfiguration.IPGatewayAddress);
success = WifiNetworkHelper.ConnectFixAddress(MySsid, MyPassword, IPConfiguration, token: cs.Token);
`