nanoframework/Home #736: Debug.IsAttached always returns False

hbraasch, 2021-05-01:
### Details about Problem

**nanoFramework area:** ( Visual Studio extension )

**VS version:** Visual Studio Community 2019 Version 16.9.4

**VS extension version:** 2019.5.0.43

**Target:** ESP32 - could be others too

**Device capabilities output:**

System Information
HAL build info: nanoFramework running @ ESP32
  Target:   ESP32_WROOM_32
  Platform: ESP32

Firmware build Info:
  Date:        Apr 29 2021
  Type:        MinSizeRel build with IDF v3.3.1
  CLR Version: 1.6.2.139
  Compiler:    GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  DebugIsAttachedTester, 1.0.0.0
  mscorlib, 1.10.3.0

Native Assemblies:
  mscorlib v100.5.0.6, checksum 0x7B586F51
  nanoFramework.Runtime.Native v100.0.8.0, checksum 0x2307A8F3
  nanoFramework.Hardware.Esp32 v100.0.7.2, checksum 0x1B75B894
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.4, checksum 0x39DA4F21
  System.Net v100.1.3.4, checksum 0x76BCAD4E
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x84A8FB94
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.4.2, checksum 0x48031DC5
  Windows.Devices.Wifi v100.0.6.2, checksum 0xA94A849E
  Windows.Storage v100.0.2.0, checksum 0x5160A7B6

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe436c  0x0001bc00
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x180000     nanoCLR
      1    0x00190000       1      0x230000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x180000 (1536kB)   nanoCLR
  0x00190000    0x230000 (2240kB)   Deployment

### Description

The instruction Debug.IsAttached always returns False (when ran under debugger or straight after reboot when attaching serial monitor (no debugger))

### Detailed repro steps so we can see the same problem

1. Create default app. Add line:  Debug.WriteLine($"Debugger IsAttached =: {Debugger.IsAttached}");

2. Run app with debugger. See that trace always reports "Debugger IsAttached =: False

3. Sample app zip file attached.

### Expected behaviour
Trace should report "Debugger IsAttached =: True when running under debugger.
Trace should report "Debugger IsAttached =: False when running after reboot (attach serial monitor to witness trace)

### Screenshot
[image]
Arrow 1 shows the debugger is attached (breakpoint being hit)
Arrow 2 shows the instruction af fault and its outcome

Zip file:
[DebugIsAttachedTester.zip](https://github.com/nanoframework/Home/files/6408710/DebugIsAttachedTester.zip)

josesimoes, 2021-05-13:
Could you please confirm if this is still happening with the latest changes in Wire Protocol?

hbraasch, 2021-05-13:
I can confirm it now works correctly. Tested on ESP32_WROOM_32. Well done!!

Below is a run without debugger... (False)
[image]

Below is a run with debugger... (True)
[image]

Regards

closed 2021-05-13 by josesimoes