nanoframework/Home #738: UnitTestLauncher using Debug.WriteLine

edleno2, 2021-05-04:
### Details about Problem

**nanoFramework area:** nanoFramework.TestFramework

### Description

The UnitTestLauncher is using Debug.WriteLine instead of Console.WriteLine.  Release versions of the tool will not generate output to the proper Visual Studio window.
### Detailed repro steps so we can see the same problem

1. (to be validated - there will be missing messages, will update this after testing)

2.

...

### Other suggested things

### Expected behaviour

Following displays should now appear in Visual Studio when using a Release version of the tool:
Debug.WriteLine($"Test passed: {method.Name}, {totalTicks}");
Debug.WriteLine($"Test failed: {method.Name}, {ex.Message}");

#### Sample Project

### Screenshot

### Additional context

Similar to problem that we fixed with the logging framework

### I've made an effort to fix the bug

josesimoes, 2021-05-13:
Don't want to look like I'm pushing you to, but when do you think you can tackle this one?
There is a hackaton next week and would like to have the test framework published with this one fixed... 😉

edleno2, 2021-05-13:
> Don't want to look like I'm pushing you to, but when do you think you can tackle this one?
> There is a hackaton next week and would like to have the test framework published with this one fixed... 😉

No problem.  I've already made the change, just had to figure out how to test the launcher, which I've done.  So I'll run some tests with debug and release versions and submit a PR.  Should be done by end-of-day.

SandorDobos, 2021-05-14:
Not sure but Debug.WriteLine by definition should not do anything in release mode. Quick search on google confirms this. At least a #define DEBUG should be present somewhere to make it write something. I didnt checked that our implementation honors DEBUG presence, but I am sure that simply enabling Debug.WriteLine in release builds wont be the good way :)

Ellerbach, 2021-05-14:
In .NET, when building, all Debug.Write/WriteLine are just removed at compilation. So to use the UnitTestLauncher, you have to use it in Debug mode, that's, so far, the only way.
The nuget uses the Debug version of the UnitTestLauncher.
Yes, Console should be better long term but so far, it's not present anymore in nano. That should be added back for those scenarios. In the meantime, use Debug as compile mode for the UnitTestLauncher.

josesimoes, 2021-05-14:
@SandorDobos yes `Debug.Writeline` is decorated with the proper attributes to have it working as expected.

The goal here was to replace it with `Console.WriteLine` which is now internal.

@Ellerbach `Console.WriteLine` is there but's internal to use in situations like these. 😉

closed 2021-05-14 by josesimoes