nanoframework/Home #753: nanoff command unable to find ESP32_WROOM_32 firmware in repository

Kleebz, 2021-05-16:
Details:
When attempting to flash the firmware, Error  code E9005 is displayed. 
"Error E9005: Can't find the target in the Bintray repository."

Steps to reproduce:
1. Open the Package Manager Console in Visual Studio 2019
2. Install nanoFirmwareFlasher : "dotnet tool install -g nanoFirmwareFlasher"
3. Update nanoFirmwareFlasher  : "dotnet tool update -g nanoFirmwareFlasher"
4. Attempt to install the firmware : "nanoff --update --target ESP32_WROOM_32 --serialport COM3 --stable -v diag" 
[image]

josesimoes, 2021-05-16:
You're using an outdate version of nanoff. Please update.

closed 2021-05-16 by josesimoes

AbhiAbzs, 2021-05-17:
I was also facing the same issue. The error message was
```
Trying to find ESP32_WROOM_32 in community targets repository...Error E9005
```
I have tired updating as well but that also didn't worked.
The whole confusion with updating the _nanoFramework_ Flasher was due to the fact that the earlier method for installing & updating the nanoFramework flashed was
```pwsh
dotnet tool install -g nanoFirmwareFlasher
dotnet tool update -g nanoFirmwareFlasher
```
using which I was getting this issue
[image]
[image]

I just uninstalled that using bellow command & that installed the _nanoFramework_ using the next command which was from its documentation & then it just worked (you would be required to press the boot button).
```pwsh
dotnet tool uninstall -g nanoFirmwareFlasher
dotnet tool install -g nanoff
```
[image]
Press boot button on esp32 board if you receive this kind of error.
[image]

Just thought of adding this for the sake of reference if any body else gets stuck. 😅

Kleebz, 2021-05-17:
@AbhiAbzs ,

This is very helpful. Thank you for taking the time to share.