nanoframework/Home #761: Port name mismatch on System.IO.Port

up-streamer, 2021-06-02:
### Details about Problem

**nanoFramework area:** (C# code )

**VS version:** (2019 )

**VS extension version:**

**Target: ESP32**

### Description
It is not possible to declare SerialPort as shown on screen shot. Please note "portName" on the Insert window and "PortName" on the code.

### Detailed repro steps so we can see the same problem

Try declare SerialPort as per screen shot.

### Other suggested things

### Expected behaviour
It should be possible to declare port name in both cases. In ( ) or as a parameter list. 

### Screenshot
[image]

Ellerbach, 2021-06-02:
That is on purpose. There is no constructor taking no portName argument.

josesimoes, 2021-06-02:
Absolutely! Unlike the full .NET API, we don't have the constructor _without_ the COM port name. We have to keep things simple and this is one of those choices. 😉

closed 2021-06-02 by josesimoes

up-streamer, 2021-06-04:
Understood.
It is not a real issue anyway... 
Thanks.