nanoframework/Home #762: System.IO.Port watchChar declaration causes exception

up-streamer, 2021-06-02:
### Details about Problem

**nanoFramework area:** (C# code )

**VS version:** ( 2019 )

**VS extension version:**

**Target: ESP32**

**Device capabilities output:**

### Description
If WatchChar is used during port declaration it will cause System.Object.Dispose.Exception.

### Detailed repro steps so we can see the same problem

Declare a serial port and add a watch char on declaration, the exception will happen soon after deployed.

### Expected behaviour
Declaration of any  WatchChar character should not cause this exception.

### Screenshot
[image]

Ellerbach, 2021-06-02:
The port needs to be open first. You can't setup the WatchChar before

josesimoes, 2021-06-02:
@Ellerbach technicaly there is no reason for that to happen, right? I mean, just like with all the other properties, there is no harm on being able to set this one before opening the port.

Ellerbach, 2021-06-02:
> @Ellerbach technicaly there is no reason for that to happen, right? I mean, just like with all the other properties, there is no harm on being able to set this one before opening the port.

A "NativeInit/NativeConfig" needs to happen to be able to set the WatchChar so the UART is initialized. I guess it's fixable but adding a NativeConfig call before setting the WatchChar should just work.

josesimoes, 2021-06-26:
Fixed in nanoframework/System.IO.Ports#16.

closed 2021-06-26 by josesimoes