nanoframework/Home #764: Removing BME280 from nf-Community-Contribution as duplicate with nanoFramework.IoT.Device

Ellerbach, 2021-06-02:
### Details about Problem

**nanoFramework area:**  C# code 

### Description

Before removing BME280 from nf-Community-Contribution as duplicate with nanoFramework.IoT.Device, I would like to get the feedback from its original author @vgolovanov

### Detailed repro steps so we can see the same problem

Original contribution: https://github.com/nanoframework/nf-Community-Contributions/tree/master/drivers/BME280
Most recent contribution: https://github.com/nanoframework/nanoFramework.IoT.Device/tree/develop/devices/Bmxx80

### Other suggested things

### Expected behaviour

#### Sample Project

### Screenshot

### Additional context

To avoid duplication of code and allowing a better maintenance. Newer version supports a larger family and is unit agnostic using UnitsNet.

### I've made an effort to fix the bug

vgolovanov, 2021-06-02:
I think it's just fine to remove it. I used that library for a sample weather station. But the original code exists in the MBNSoftware repo.

josesimoes, 2021-07-08:
Migration completed. Device driver is now at https://github.com/nanoframework/nanoFramework.IoT.Device/tree/develop/devices/Bmxx80

closed 2021-07-08 by josesimoes