nanoframework/Home #765: Removing IO-Expander_MCP23017 from nf-Community-Contribution as present in nanoFramework.IoT.Device

Ellerbach, 2021-06-02:
### Details about Problem

**nanoFramework area:** C# code

### Description

- Removing IO-Expander_MCP23017 from nf-Community-Contribution as present in nanoFramework.IoT.Device
- Larger family in the new repository.

Tagging @LordAsdi as original author.

### Other suggested things

### Expected behaviour

#### Sample Project

### Screenshot

### Additional context

### I've made an effort to fix the bug

josesimoes, 2021-08-18:
@Ellerbach hasn't this been moved already?

networkfusion, 2021-09-10:
I think so: https://github.com/nanoframework/nf-Community-Contributions/commit/bb3d0e7ae2373a22470bd69448001cfab23d50e9 . Given that, I am closing... Feel free to reopen if not.

closed 2021-09-10 by networkfusion