nanoframework/Home #769: String.Format() and Double.ToString() create string with a zeroed value when adjusting the precision

pikrog, 2021-06-03:
### Details about Problem
**nanoFramework area:** C# code
### Description
Neither `String.Format()` nor `Double.ToString()` format floating point numbers properly. Attempting to trim digits after the dot will result in a zeroed value. For example, formatting a double that holds a value of `25.4` with parameter `0.0` results in a string: `"0.0"`. A workaround to this is to use the `F` parameter, for example `F1`.
### Detailed repro steps so we can see the same problem
Just write this line of code:
`Debug.WriteLine("Temperature: " + (25.4).ToString("0.0") + " °C");`
or:
`Debug.WriteLine(String.Format("Temperature: {0:0.0} °C", 25.4));`
and check the output.
### Expected behaviour
Output:
`Temperature: 25.4 °C`

edleno2, 2021-06-07:
@josesimoes - since we decided to document the string formats should we just close this issue?  It is working as intended at this point I believe.

josesimoes, 2021-06-07:
@edleno2 sure! When that explanation is there, please add a link to it here for completion.