nanoframework/Home #770: Add explanation on supported numeric format strings supported

josesimoes, 2021-06-04:
### Details about Problem

**nanoFramework area:** ( Docs )

### Description

- Need to add a short explanation of the supported numeric format strings to the docs. Possibly [here](https://docs.nanoframework.net/content/architecture/simplifications-and-trade-offs.html>.

edleno2, 2021-06-06:
Also adding some String.Format examples to the CoreLibrary unit tests to match this documentation.

edleno2, 2021-06-07:
Just an update: I have many/most of the test cases written.  So far found three bugs in either String.Format or [unit].Parse causing test failures.  I'll be including a PR for nanoCLR to fix those issues along with the test cases in CoreLibrary, and the matching documentation.

edleno2, 2021-06-07:
@josesimoes - do you want me to create a separate issue for bug fixes and new unit tests in nf-interpreter, or can I combine this documentation change with code changes?

josesimoes, 2021-06-07:
> @josesimoes - do you want me to create a separate issue for bug fixes and new unit tests in nf-interpreter, or can I combine this documentation change with code changes?

No need for more related issues. Let's keep this simple.
Please submit PRs for each on those. Always have "atomic" PRs as much as possible.

closed 2021-06-16 by josesimoes