nanoframework/Home #771: System.IO.Port event handler stops after when read all characters from buffer

up-streamer, 2021-06-04:
### Details about Problem

**nanoFramework area:** (C# code )

**VS version:** ( 2019 )

**Target: ESP32**

**Device capabilities output:**

### Description
When trying to read a stream of 4 bytes every 100 milliseconds, 32 characters is stored in the buffer.
No new character is added on buffer from read stream.
When the available character reaches zero the program finishes, writing "Done." on debug window

### Detailed repro steps so we can see the same problem

1. A stream of data is necessary, same amount of bytes and fixed interval around 100ms. Already sending data prior booting.

2. Read data from stream and print on debug window.

### Expected behaviour
Debug window should kept printing data endlessly.

#### Sample Project
```C#

SerialDevice = new SerialPort(port);
		        // set parameters
			SerialDevice.BaudRate = 9600;
			SerialDevice.Parity = Parity.None;
			SerialDevice.StopBits = StopBits.One;
			SerialDevice.Handshake = Handshake.None;
			SerialDevice.DataBits = 8;
			SerialDevice.Mode = SerialMode.Normal;
                        SerialDevice.DataReceived += printData;
.
.
.
		int count = 0;
		byte [] dataauto = new byte[4];

		/// <summary>
		/// Serial data received event
		/// </summary>
		/// <param name="Sender"></param>
		/// <param name="EventData"></param>

		private void printData(object sender, SerialDataReceivedEventArgs e)
		{
			count++;
			Debug.WriteLine(count.ToString());
			if (SerialDevice.BytesToRead > 0)
			{
				Debug.WriteLine("Bytes to Read = " + SerialDevice.BytesToRead.ToString());
				var dt = SerialDevice.Read(dataauto, 0, 4);
				Debug.WriteLine(" Bytes read = " + dt);
				Debug.WriteLine($"RX = {(byte)dataauto[0] + " " + (byte)dataauto[1] + " " + (byte)dataauto[2] + " " + (byte)dataauto[3]}");
			}
```

### Additional context
Output of code above:
The thread '<No Name>' (0x1) has exited with code 0 (0x0).
1
Bytes to Read = 32
 Bytes read = 4
RX = 255 1 10 11
2
Bytes to Read = 28
 Bytes read = 4
RX = 255 1 14 15
3
Bytes to Read = 28
 Bytes read = 4
RX = 255 1 14 15
4
Bytes to Read = 24
 Bytes read = 4
RX = 255 1 10 11
5
Bytes to Read = 20
 Bytes read = 4
RX = 255 1 10 11
6
Bytes to Read = 16
 Bytes read = 4
RX = 255 1 14 15
7
Bytes to Read = 12
 Bytes read = 4
RX = 255 1 14 15
8
Bytes to Read = 8
 Bytes read = 4
RX = 255 1 10 11
9
Bytes to Read = 4
 Bytes read = 4
RX = 255 1 14 15
Done.
Waiting for debug commands...
The program '[1] .NET nanoFramework application: Managed' has exited with code 0 (0x0).

josesimoes, 2021-08-11:
@up-streamer can you confirm if you still are seeing this issue?

up-streamer, 2021-08-15:
@josesimoes 
I updated firmware, nugget packages and vs extension and have not seeing the issue since.
I think it was caused by some version mismatch.

josesimoes, 2021-08-15:
Closing issue as it seems to be gone.

closed 2021-08-15 by josesimoes