nanoframework/Home #772: ToString() fails with null object

josesimoes, 2021-06-04:
### Details about Problem

**nanoFramework area:** (mscorlib)

### Description
Calling ToString() with a null object throws an exceptio.

### Detailed repro steps so we can see the same problem

	object iAmNull = null;
	Debug.WriteLine($"This is supposed to work: {iAmNull}");

edleno2, 2021-06-07:
This may have come from my cleanup of unit tests.  I'll include testing it along with the String.Format issue, but submit a separate PR just for this issue.   I had to test on .net core - didn't think this would work this way!

edleno2, 2021-06-07:
I believe it's not ToString() but String.Format that is supposed to accept null arguments.  I do remember changing that behavior so I'll have to go back to which cases should work and which should get a null exception.

edleno2, 2021-06-07:
Waiting for checks to complete.  I'll send for review tomorrow morning if all of the checks pass.

closed 2021-06-07 by josesimoes