nanoframework/Home #779: Contains("<<Something") doesn't return true while it should

Ellerbach, 2021-06-18:
### Details about Problem

**nanoFramework area:** C# code

**VS version:** (2017 | 2019 )

**VS extension version:**

**Target:**

**Device capabilities output:**

### Description

Issue with string.Contains for a specific character string.

### Detailed repro steps so we can see the same problem

```csharp
const string C9PatternMainStyle = "<<Main>$>g__";
string mtName = "<<Main>$>g__MethodCalbackTest|0_3";
if (mtName.Contains(C9PatternMainStyle))
{
  // We should arrive here but Contains doesn't find it!
}
```

Notes:

- Works: `mtName.Contains("Main>$>"))`
- Does not work: `mtName.Contains("<<Main"))`

### Other suggested things

### Expected behaviour

#### Sample Project

### Screenshot

### Additional context

### I've made an effort to fix the bug

josesimoes, 2021-06-18:
Our API currently has no `Contains(...)` and friends. We should look into adding these. 
Most of them can (are) pure C# code, so they won't "cost" anything in native size increase.

Ellerbach, 2021-06-18:
You're right :-) My bad :-D It's an extension I have and the issue is in the extension! Found my issue.

josesimoes, 2021-06-18:
Almost ready. Just adding a pile of unit tests 😜

closed 2021-06-19 by josesimoes