nanoframework/Home #782: Support for generics T<>

josesimoes, 2021-06-23:
**nanoFramework area:** (CLR)

We are actively working on adding support for generics.

- [x] CLR
- [x] Type System
- [x] Interpreter
- [x] Debugger Library
- [x] Metadata processor
- [ ] Support in Visual Studio debugger engine

acrigney, 2021-09-24:
Could we have an update on your status on generics support please? I think this feature is a show stopper for us.

josesimoes, 2021-09-24:
> Could we have an update on your status on generics support please? I think this feature is a show stopper for us.

Hey @acrigney! Thanks for asking. This is it's still on the queue waiting for free time to get back to it...

Martinn2, 2022-02-04:
Any news about this issue?

Ellerbach, 2022-02-05:
> Any news about this issue?

This will require funding, more than what is available as it's still a long road to go and hard to do just as a community effort.

DocBrown101, 2022-02-07:
To be able to sort the whole thing a little better, since I unfortunately could not find a Roadmap.
Are the "generics" needed first before we can use ".NET standard" libraries?

Ellerbach, 2022-02-07:
> Are the "generics" needed first before we can use ".NET standard" libraries?

Short answer: Yes
Longer answer: you need more to support .NET Standard. Also, there is technically many ways to support .NE Standard even without generics or async or Linq. But that would not help much :-) In all cases, there is a very very very long way to support .NET Standard on such small MCU!

josesimoes, 2022-02-07:
> To be able to sort the whole thing a little better, since I unfortunately could not find a Roadmap. Are the "generics" needed first before we can use ".NET standard" libraries?

The existing CLR and execution engine require a "parsed" version of a .NET library. That's why we can't reference directly the "standard" NuGets and .NET libraries.
Main reason being that you won't have enough flash and RAM to deploy them in the 1st place.
Generics or no generics that it's unlikely to change.
So you'll always need a trimmed down version of any .NET library.

DocBrown101, 2022-02-07:
Okay, that's very unfortunate.
It sounds more like it will never be possible in the future because there are not enough resources for .NET in the micro-controller area.

josesimoes, 2022-02-07:
@DocBrown101 not sure if I follow up... we are running .NET on microcontrollers for years now!
Just that you can't use the same libraries for the reasons explained.

DocBrown101, 2022-02-07:
Sorry, that was worded wrong, of course I meant ".NET Standard"

josesimoes, 2022-02-07:
The same applies. .NET nanoFramework mscorlib (along with the other libs that "mirror" .NET) are aligned with the full framework as much as possible. The exceptions are, obviously, the lack of generics and all that goes in tow like async/await, etc.
When generics become available the "alignment" will be worked on.

t9mike, 2022-02-08:
Any estimate on funding required? What about a Gofundme? Get some heavy hitters in the .NET world to help spread the word (I could hit up one). I think a lot of C# fans would throw in some $ as they like the idea of improvement in this area even if they are not embedded developers or hobbyists.

@Ellerbach said "still a long road to go" but 5 of 6 tasks are complete. Is that final task super challenging?

Ellerbach, 2022-02-09:
Hi @t9mike, 
It’s hard to estimate how much funding is required but we are speaking in units of 10K€, most likely 40K€+. Even if you see checked boxes, there is still quite some things to do. A checkbox in this context means: the minimum to see if it’s working is done. It doesn’t mean, it’s production code quality ready with all the tests. As soon as you introduce something that is a large change like this one, there will be, some small issues which will have to be investigated, some may be edge cases because some elements may be missing in the platform and will have to be added. It took years to have something stable on generics on the .NET Framework! OK, the compiler is now perfect for that but it has to be fitting into the small device.
The debugger part is extremely complex. It takes hours to setup (2 Visual Studio sessions plus 1 VS Code plus hardware debugging), you need then to understand step by step what’s happening, it’s a lot of work.
I love the idea of having a fund raising, I would love as well to see professional users funding this seriously as well. I’m sure that all together, something can be achieved. And it has as well to be fair and enough to make the changes and support afterward properly. Up to the point it can become a community effort.

lvzuoxin, 2022-08-10:
Support for generics <T> now?
I want use nanoCLR for game develop C# script

alberk8, 2022-08-10:
> Support for generics now? I want use nanoCLR for game develop C# script

I would think it is easier and cheaper to develop your game in Raspberry Pi and full dotnet framework is available.  Maybe look at [Meadows](https://www.wildernesslabs.co/) where full framework C# is supported.

josesimoes, 2022-08-10:
Development of this it's on hold. Progress will be reported here. 🙂

acrigney, 2022-08-10:
Meadows is awesome with full generics support.

DocBrown101, 2022-08-10:
@acrigney But Meadow only works with a single board, which isn't cheap either.

28106830, 2025-01-08:
I sincerely hope to support generics. Currently, many MCUs have a clock frequency greater than 200MHz and a storage capacity of 16MB, RAM8MB
I hope to configure generics as optional, so that high-capacity microcontrollers can use this benefit

mikeobrien, 2025-12-15:
This is HUGE! Big thanks to everyone involved!