nanoframework/Home #785: stm32f091 debug doesn't work

subsneer, 2021-07-03:
### Debug output and breakpoints doesn't work

**nanoFramework area:** Visual Studio extension

**VS version:** VS Community 2017 15.9.36

**VS extension version:** 2017.6.0.54

**Target:** stm32f091

**Firmware image version:**
  Date:        May  7 2021
  Type:        MinSizeRel build with ChibiOS v20.3.3.3
  CLR Version: 1.6.4.188
  Compiler:    GNU ARM GCC v9.3.1

**Device capabilities output:**
14:10:28.363 [Starting debug engine on nanoDevice]
14:10:28.376 [Connecting to debug engine...OK]
14:10:28.562 [Assemblies to deploy:]
14:10:28.564 [Assembly: NFApp1 v1.0.0.0 (336 bytes)]
14:10:28.612 [Assembly: mscorlib v1.10.3.0 (31220 bytes)]
14:10:28.613 [Deploying 2 assemblies to device]
14:10:29.410 [Incrementally deploying assemblies to the device]
14:10:29.411 [Erasing block @ 0x0802A000...]
14:10:30.244 [Erasing block @ 0x0802A800...]
14:10:31.032 [Erasing block @ 0x0802B000...]
14:10:31.894 [Erasing block @ 0x0802B800...]
14:10:32.596 [Erasing block @ 0x0802C000...]
14:10:33.306 [Erasing block @ 0x0802C800...]
14:10:34.184 [Erasing block @ 0x0802D000...]
14:10:35.062 [Erasing block @ 0x0802D800...]
14:10:35.707 [Erasing block @ 0x0802E000...]
14:10:36.662 [Erasing block @ 0x0802E800...]
14:10:37.243 [Erasing block @ 0x0802F000...]
14:10:38.061 [Erasing block @ 0x0802F800...]
14:10:38.962 [Erasing block @ 0x08030000...]
14:10:39.769 [Erasing block @ 0x08030800...]
14:10:40.389 [Erasing block @ 0x08031000...]
14:10:41.157 [Erasing block @ 0x08031800...]
14:10:41.610 [Assemblies successfully deployed to the device]
14:10:41.981 [Attempting to connect debugger engine (1/5)]
14:10:41.981 [Assigning debug engine of target device]
14:10:41.981 [Device running nanoCLR, updating debug flags...]
14:10:42.809 [Attempting to connect debugger engine (1/5)]
14:10:42.809 [Device running nanoCLR, updating debug flags...]
14:10:43.677 [Device is running CLR, requesting reboot and pause for debugger (1/5).]
14:10:43.957 [Reboot execution succesfull.]
14:10:44.458 [Attempting to connect debugger engine (1/5)]
14:10:44.458 [Device running nanoCLR, updating debug flags...]
14:10:45.252 [Device is running the CLR in initialized state]
14:10:45.252 [The debugger engine is listening to debug events from the nanoDevice target]
14:10:46.655 [Updating information about assemblies on the debugging target.]
14:10:48.940 [Error loading assemblies. Assemblies count is 0.]

### Description
I can’t get step debug working on it, Breakpoints doesn’t work as well.
I tried default empty project with Debug.WriteLine and Blinky examples and looks like deployment is OK cause led is blinking but no debug or debug output there.
There is one error at the end of .NET nanoframework extension output:

### Detailed repro steps so we can see the same problem

1. Build default nano project
2. Set breakpoint at Debug.WriteLine("Hello from nanoFramework!");
3. F5

### Other suggested things

### Expected behaviour

Waiting for stop at breakpoint or "Hello from nanoFramework!" in debug output

### Screenshot

### Additional context

I've seen same behavior with VS2019

### Make an effort to fix the bug

josesimoes, 2021-07-05:
@subsneer what you have above as device capabilities, isn't correct. Please click this icon here in device explorer window
[image]

Also make sure you have installed the latest nanoff version.

The version that's being reported above it's not the most recent one, neither stable nor preview.

subsneer, 2021-07-05:
OK, so I check it with VS2019 now.
I updated nanoff to latest one. It was 1.22.4 and now it's 1.22.5. And I tried to get latest preview update - ST_NUCLEO64_F091RC-1.6.5-preview.413. But unfortunately after this version was installed Device explorer couldn't find stm board any more. I check it couple of times and have to rollback to ST_NUCLEO64_F091RC-1.6.4.5.
There is f/w installation result:
_Trying to find ST_NUCLEO64_F091RC in stable repository...OK
Downloading firmware package...OK
Extracting ST_NUCLEO64_F091RC-1.6.4.5.zip...OK
Updating to 1.6.4.5
Connected to JTAG device with ID 066CFF514855657867184743
Flashing device...
nanoBooter.hex
nanoCLR.hex
Flashing completed...
Reset MCU on device... OK_

There is device info:

System Information
HAL build info: nanoCLR running @ ST_NUCLEO64_F091RC
  Target:   ST_NUCLEO64_F091RC
  Platform: STM32F0

Firmware build Info:
  Date:        May  7 2021
  Type:        MinSizeRel build with ChibiOS v20.3.3.3
  CLR Version: 1.6.4.188
  Compiler:    GNU ARM GCC v9.3.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: YES
  nanoBooter: v1.6.4.188
  IFU capable: NO
  Has proprietary bootloader: NO

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.6, checksum 0x7B586F51
  nanoFramework.Runtime.Native v100.0.8.0, checksum 0x2307A8F3
  nanoFramework.Hardware.Stm32 v100.0.4.4, checksum 0x0874B6FE
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x84A8FB94
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.0.0, checksum 0x48031DC5

I don’t know if it is OK, cause version is 1.6.4.5 but CLR Version: 1.6.4.188, but debug still doesn’t work and there is still same output:

11:26:03.020 [Starting debug engine on nanoDevice]
11:26:03.038 [Connecting to debug engine...OK]
11:26:03.395 [Assemblies to deploy:]
11:26:03.395 [Assembly: Blinky_2019 v1.0.0.0 (340 bytes)]
11:26:03.441 [Assembly: nanoFramework.Runtime.Events v1.9.0.0 (3412 bytes)]
11:26:03.456 [Assembly: mscorlib v1.10.3.0 (31220 bytes)]
11:26:03.457 [Assembly: Windows.Devices.Gpio v1.5.2.0 (5424 bytes)]
11:26:03.457 [Deploying 4 assemblies to device]
11:26:04.304 [Incrementally deploying assemblies to the device]
11:26:04.304 [Erasing block @ 0x0802A000...]
11:26:05.209 [Erasing block @ 0x0802A800...]
11:26:05.845 [Erasing block @ 0x0802B000...]
11:26:06.662 [Erasing block @ 0x0802B800...]
11:26:07.419 [Erasing block @ 0x0802C000...]
11:26:08.208 [Erasing block @ 0x0802C800...]
11:26:09.213 [Erasing block @ 0x0802D000...]
11:26:09.985 [Erasing block @ 0x0802D800...]
11:26:10.599 [Erasing block @ 0x0802E000...]
11:26:11.438 [Erasing block @ 0x0802E800...]
11:26:12.321 [Erasing block @ 0x0802F000...]
11:26:12.944 [Erasing block @ 0x0802F800...]
11:26:13.842 [Erasing block @ 0x08030000...]
11:26:14.537 [Erasing block @ 0x08030800...]
11:26:15.321 [Erasing block @ 0x08031000...]
11:26:16.138 [Erasing block @ 0x08031800...]
11:26:16.922 [Erasing block @ 0x08032000...]
11:26:17.799 [Erasing block @ 0x08032800...]
11:26:18.559 [Erasing block @ 0x08033000...]
11:26:19.393 [Erasing block @ 0x08033800...]
11:26:19.891 [Assemblies successfully deployed to the device]
11:26:20.143 [Attempting to connect debugger engine (1/5)]
11:26:20.143 [Assigning debug engine of target device]
11:26:20.143 [Device running nanoCLR, updating debug flags...]
11:26:21.034 [Attempting to connect debugger engine (1/5)]
11:26:21.034 [Device running nanoCLR, updating debug flags...]
11:26:21.895 [Device is running CLR, requesting reboot and pause for debugger (1/5).]
11:26:22.175 [Reboot execution succesfull.]
11:26:22.676 [Attempting to connect debugger engine (1/5)]
11:26:22.676 [Device running nanoCLR, updating debug flags...]
11:26:23.469 [Device is running the CLR in initialized state]
11:26:23.469 [The debugger engine is listening to debug events from the nanoDevice target]
11:26:24.852 [Updating information about assemblies on the debugging target.]
11:26:29.340 [Error loading assemblies. Assemblies count is 0.]

closed 2021-08-16 by josesimoes